blue-systems/plamo-initial-dev_DEPRECATED #155: Add new dependencies to plasma-settings

mck182, 2015-07-15:
On Harald's request, a new issue.

New required deps for plasma-settings
 * libaccounts-qt5
 * libsignon-qt5
 * libkaccounts 
 * libkgapi/frameworks
 * kcontacts

Runtime deps
 * kaccounts-integration
 * kaccounts-providers
 * signon-ui (the patched version from #153)
 * signond

apachelogger, 2015-07-15:
buil depends adjusted, signon stuff waiting for signon-ui patchery

mck182, 2015-07-15:
Turns out plasma-settings will require KF5CalendarCore as a build dependency too; it's a find_dependency of kgapi.

http://mobile.kci.pangea.pub:8080/job/vivid_unstable_plasma-settings-packaging_bin_amd64/18/console

apachelogger, 2015-07-16:
thought I did that

apachelogger, 2015-07-16:
ah, settings wasn't triggered with the fix. triggered it. should be fixed now

apachelogger, 2015-07-16:
as it turns out reading helps and this is really #158 as the package is there the versions just dont match

apachelogger, 2015-07-16:
Dependency wise this should be sorted as kaccounts-integraton drags in the relevant parts of the signon stack (among them "signon-ui-x11, signond")

closed 2015-07-16 by apachelogger

jriddell, 2015-07-16:
<mck182> sebas: yup, the module is now built only if kaccounts && libaccounts-qt5 && libsignon-qt5 && libkgapi && libkcontacts have been found