blue-systems/plamo-initial-dev_DEPRECATED #160: [apps]: funny app for testing sound

star-buck, 2015-07-16:
ship this click app for fun and testing sound:
https://www.dropbox.com/s/vd1swwzbxaj0ihw/com.popey.stallboard_0.2_all.click?dl=0

star-buck, 2015-07-31:
not shipped as of IMG137

star-buck, 2015-11-15:
@bhush9 : please include this default for now as testing app for 
* sound and 
* to see if click/mir apps still work

bhush9, 2015-11-16:
Turns out it is already installed, just not showing up in launcher area.. will investigate

bhush9, 2015-11-16:
Actually it is not getting installed at all..

```
15:00:09 + pkcon install-local file.click --allow-untrusted
15:00:09 Failed to contact PackageKit: Could not connect: No such file or directory
```

star-buck, 2015-11-16:
The file on dropbox is there?
Have you followed this tutorial from @aleixpol : https://github.com/blue-systems/pm-img/issues/21#issuecomment-112996191 ?

star-buck, 2015-12-16:
@bhush9 @aleixpol : please look into make that .click-app install again

aleixpol, 2015-12-21:
For starters `packagekit-plugin-click` package is needed:
```
$ phablet@ubuntu-phablet:~$ pkcon install-local com.popey.stallboard_0.2_all.click
Installing files              [=========================]         _0.2_all.click 
Finished                      [=========================]         
Fatal error: MIME type 'application/x-click' not supported /home/phablet/com.popey.stallboard_0.2_all.click
$ sudo apt install packagekit-plugin-click
...
$ phablet@ubuntu-phablet:~$ pkcon install-local com.popey.stallboard_0.2_all.click
WARNING:root:debsig-verify not available; cannot check signatures
Cannot install /home/phablet/com.popey.stallboard_0.2_all.click: Framework "ubuntu-sdk-13.10" not present on system (use --force-missing-framework option to override)
```

This unveils another problem I'm not sure how to fix. Anybody knows what a `Framework "ubuntu-sdk-13.10"` is?

star-buck, 2015-12-21:
Did we have this before while it was working?

aleixpol, 2015-12-21:
I don't think I ever saw this error message.

star-buck, 2015-12-21:
So the question is: Did we ship it previously?

bhush9, 2016-01-05:
[image]

I got them working in theory, but requires some proper changes to get them working on mobile IMG.

star-buck, 2016-01-05:
What changes exactly are needed?

bhush9, 2016-01-06:
Mainly _something_ that makes them show up in our launcher, and instead of using pkcon using click command to install them.

That _something_ is using http://click.readthedocs.org/en/latest/hooks.html

aleixpol, 2016-01-07:
Last time I checked, they were appearing in the menu. Do you know what changed, @bhush9?

bhush9, 2016-01-07:
click apps that are shipped as deb do appear in menu fine but .click apps doesn't.

star-buck, 2016-01-07:
Deb files are just that: .deb (not click?).
They all maybe ubuntu touch apps, but click is the package format?

bhush9, 2016-01-07:
not sure I understood question?

bhush9, 2016-01-07:
Ubuntu apps are either deb packages or .click packages.. deb packages gets shown into launcher as they ship desktop files in /usr/share/applications however .click applications are installed into /opt/click.ubuntu.com/ and don't install .desktop files where our launcher searches them. so they don't appear in our launcher

star-buck, 2016-01-07:
Now you got it right :)

star-buck, 2016-02-23:
We can also get official click apps from ubuntu from here:
https://core-apps-jenkins.ubuntu.com/job/filemanager-app-ci/

bhush9, 2016-02-24:
rather more proper place : http://people.canonical.com/~ubuntu-archive/click_packages/

star-buck, 2016-02-24:
nice!
@bhush9 : please try installing one of those click apps on latest xenial img:
e.g. http://people.canonical.com/~ubuntu-archive/click_packages/com.ubuntu.sudoku_1.1.382_all.click

bhush9, 2016-02-24:
Installing click apps is not a problem, there are multiple click packages installed already.. see `click list`. Problem is we have no real way to show them in launcher and launch them.

star-buck, 2016-02-24:
@notmart : any way to make installed .click apps icons appear in our launcher list?

bhush9, 2016-02-26:
I've put together script in plasma-phone-dev-setup which looks for desktop files and puts them in ~/.local/share/applications with few edits.. This script takes argument of package name..

```
click-process com.popey.stallboard
```

Now it will appear in launcher list..

bhush9, 2016-03-01:
Some packages are not supposed to work at all, modified ticket above.

closed 2016-03-02 by star-buck