blue-systems/plamo-initial-dev_DEPRECATED #163: [kwallet]: disable kwallet for now (messes with setting up wifi from top panel)

star-buck, 2015-07-21:
two popups come up and you cannot close the second one...

mck182, 2015-07-21:
I have working pam-kwallet5 (not yet upstream), maybe that could be deployed on the phone?

star-buck, 2015-07-21:
mck182: i suggest to not bring annyoing kwallet on the phone for akademy, purely to avoid negative connotations whenever possible...

star-buck, 2015-07-21:
@Riddell: another conflict, so please remove signon-kwallet-extension, quoting:

(4:00:23 PM) mck182___: however it needs signon-kwallet-extension removed first, but it's a hard dependency of some other things so the packaging needs to be adjusted first

apachelogger, 2015-07-22:
the important bit is -> "[..] but it's a hard dependency of some other things so the packaging needs to be adjusted first"

star-buck, 2015-08-10:
kwallet still pops up

star-buck, 2015-09-19:
neon IMG1: popup still prevents connecting to WLAN from interface, can we disable kwallet for now, so kded5 doesnt request i...

bhush9, 2015-09-19:
As a workaround,

we could disable kwallet5 with this in .config/kwalletrc

```
[Wallet]
Close When Idle=false
Close on Screensaver=false
Enabled=false
Idle Timeout=10
Launch Manager=false
Leave Manager Open=false
Leave Open=true
Prompt on Open=false
Use One Wallet=true
```

bhush9, 2015-09-19:
I will give it try locally to see if it works.

apachelogger, 2015-09-19:
fwiw at this point all that is needed is
- have pam-kwallet on the CI
- have simplelogin's pam actually load the module
- ship pam-kwallet5 in the image

mck182, 2015-09-24:
I've tried, it won't work.

kwallet-pam complains about not having the password. Although maybe it requires a pre-set kwallet? Which I don't have because the dialog does not fit the touchscreen...

simplelogin: pam_kwallet5(sddm-autologin:auth): (null): pam_sm_authenticate
simplelogin: pam_kwallet5(sddm-autologin:auth): pam_kwallet5: Couldn't get password (it is empty)
simplelogin: pam_kwallet5(sddm-autologin:auth): pam_kwallet5: Prompt for password failed Unknown PAM error
simplelogin: pam_kwallet5(sddm-autologin:setcred): pam_kwallet5: pam_sm_setcred
simplelogin: pam_kwallet5(sddm-autologin:session): pam_kwallet5: pam_sm_open_session
simplelogin: pam_kwallet5(sddm-autologin:session): pam_kwallet5: open_session called without kwallet5_key

apachelogger, 2015-09-25:
Ah uh yeah, that actually makes sense. Kind of.

The way the module works (I think) is it asks PAM for the user password it was provided (from the login manager) and then salts that and throws it at kwalletd to use as password for the wallet.
BUT, on autologin there is no password involved so the module refuses to create a wallet on account of it being a passwordless wallet (unintentionally mind you).

So we could kill kwallet which likely will make apps angry (e.g. I recall plasma-nm refusing to store passwords for user-only connections without kwallet). Or we give pam_kwallet an override to allow passwordless wallets. Or we fix this properly and create a firststartwizard and unlocking tech so that we always auth against PAM with something. I mean, right now the phone is as insecure as it can be anyway.

mck182, 2015-09-25:
> Or we fix this properly and create a firststartwizard and unlocking tech so that we always auth against PAM with something

The phone could boot to a lockscreen which is a real lockscreen with passwords/pin/code, unlocking the screen for the first time would unlock the wallet. If user sets no password/pin/code, the wallet would be passwordless.

This would require the lockscreen setting to sync the password to the wallet though.

star-buck, 2015-10-10:
neon-img v9:
still comes up, leaving password fields empty brings up confirmation window (weak passwords) which is too long to be able to confirm

mck182, 2015-10-16:
Yeah, for the approach mentioned above we need:
 - https://git.reviewboard.kde.org/r/125641/
 - working lockscreen on the phone (I understand @bhush9 got something working)
 - preinstalled kwallet file

In the meantime, I think we should just ship a premade kwallet file with empty password. @jriddell can you look into that? You said you do something like that in Kubuntu.

star-buck, 2015-10-16:
Please lets do that for kwallet5 and kwallet4 for now. Riddell: Post both
solutions here, so we know how we did it just in case.

bhush9, 2015-11-05:
Small update: kwin_wayland got --lockscreen option, which will start lockscreen when used. It is not really secure at moment but enough for this purpose.

bhush9, 2015-11-05:
http://commits.kde.org/kwin/bacfd876fecf2df72ba38d651f637a720e108581

It will need packaging kscreenlocker repo and update in kwin.

bhush9, 2015-12-08:
http://commits.kde.org/scratch/davidedmundson/simplelogin/89328a01cbf67d917e2c7364289ad9c3fe0c501b

As lockscreen for mobile was fixed in kwin, I've now enabled the lockscreen on boot for phone.. @mck182 : now what needs to be done to do unlock-screen-to-unlock-wallet for mobile?

star-buck, 2015-12-19:
seems with lockscreen, kwallet no longer pops up as of img14...

closed 2015-12-19 by star-buck