blue-systems/plamo-initial-dev_DEPRECATED #164: [IMG]: 136 crashes and fails to come up?

star-buck, 2015-07-28:


llelectronics, 2015-07-28:
Updating with sudo apt-get update && sudo apt-get upgrade fixes this via adb shell. 
Seems to be some library mismatch or too old on this build

star-buck, 2015-07-30:
doesnt work here, upgrade is stuck at:
"Installing new version of config file /etc/init/bluetooth.conf ..."

star-buck, 2015-07-30:
dpkg --reconfigure is then stuck at:
Setting up bluez (5.23-1ubuntu1~mtrudel1build1) ...

star-buck, 2015-07-30:
@jriddell : Let's do a new IMG.

llelectronics, 2015-07-30:
Yeah star-buck you are right. I forgot to mention it. 
I just added exit 0 to the postinst Script of bluez as it kept on hanging because of something in it. 
After that it worked though. 

A new image should fix it altogether I guess.

star-buck, 2015-07-30:
@llelectronics :
even then much stuff on IMG 136 is borked:
camera doesnt work, muon dicsover still doesnt come up, still 3x okular icons on applauncher, none of the .click apps pe-installed, etc.

@jriddell :
Please see that tickets get solved and packages added.
I dont know if the initial 2.0GB of 2.0GB used limitation for 7-root partition after flashing is hindering the inclusion of new packages though?

jriddell, 2015-07-30:
new image is up now in kubuntu-mobile/devel

closed 2015-07-31 by star-buck