blue-systems/plamo-initial-dev_DEPRECATED #168: [launchers]: 2xtimes "Okular", 1x "Document Reader", 1x "Wordview Microsoft"

star-buck, 2015-07-31:
- the 2x "okular" are desktop versions, so lets remove them
- the "Wordview Microsoft also doesnt work and should be removed?

notmart, 2015-08-05:
The okular package is installing both the desktop and the qml version of the application..
it should be splitted in like 3 packages: the core library, the desktop application and the mobile application

notmart, 2015-08-05:
the packages should be:
the desktop app: what's is installed by the shell subfolder:
- /usr/bin/okular
- /usr/share/applications/org.kde.okular.desktop
- /usr/share/kxmlgui5/okular/shell.rc

the mobile app: what's installed by the mobile/app subfolder:
- /usr/share/kpackage/genericqml/org.kde.mobile.okular
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/TableOfContents.qml
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/Browser.qml
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/bookmark.svgz
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/Thumbnails.qml
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/FullScreenDelegate.qml
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/Documents.qml
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/Bookmarks.qml
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/ThumbnailsBase.qml
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/TreeDelegate.qml
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/bookmark.png
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/main.qml
- /usr/share/kpackage/genericqml/org.kde.mobile.okular/metadata.desktop
- /usr/share/applications/org.kde.mobile.okular.desktop

star-buck, 2015-09-14:
@notmart :
is the splitting part done as of IMG140?

notmart, 2015-09-14:
@jriddell : what that would be needed?
if it's challenging splitting the package, the qwidget based version could be blacklisted in the meantime

jriddell, 2015-09-16:
split done, building now in mobile kci, but when I try to run it it says
"Error: Error while loading page: file:///usr/share/kpackage/genericqml/org.kde.mobile.okular/contents/ui/Browser.qml:24 module "org.kde.plasma.mobilecomponents" is not installed"
where do I find that?

notmart, 2015-09-16:
plasma-mobile

bhush9, 2015-09-24:
Split is done I guess.. however IMG needs to include okular-mobile

jriddell, 2015-09-24:
on image 20150924.130033

star-buck, 2015-09-24:
will mark other issue "ready", closing the split here for now

closed 2015-09-24 by star-buck