blue-systems/plamo-initial-dev_DEPRECATED #171: [IMG 137]: "sudo apt-get upgrade" downgrades 16 packages

star-buck, 2015-07-31:
The following packages have been kept back:
  ubuntu-touch-session
The following packages will be upgraded:
  bluez libbluetooth3 libmission-control-plugins0 libmtpserver1 libsqlite3-0
  qtdeclarative5-ubuntu-keyboard-extensions0.1
  qtdeclarative5-ubuntu-ui-extras-browser-plugin
  qtdeclarative5-ubuntu-web-plugin qtdeclarative5-ubuntu-web-plugin-assets
  sqlite3 telepathy-mission-control-5 ubuntu-keyboard-data webapp-container
  webbrowser-app
The following packages will be DOWNGRADED:
  qml-module-qt-labs-folderlistmodel qml-module-qt-labs-settings
  qml-module-qtqml-models2 qml-module-qtqml-statemachine
  qml-module-qtquick-localstorage qml-module-qtquick-particles2
  qml-module-qtquick-window2 qml-module-qtquick-xmllistmodel
  qml-module-qtquick2 qmlscene qtdeclarative5-folderlistmodel-plugin
  qtdeclarative5-localstorage-plugin qtdeclarative5-particles-plugin
  qtdeclarative5-qtquick2-plugin qtdeclarative5-window-plugin
  qtdeclarative5-xmllistmodel-plugin
14 upgraded, 0 newly installed, 16 downgraded, 0 to remove and 1 not upgraded.
Need to get 2,668 kB of archives.
After this operation, 366 kB of additional disk space will be used.

star-buck, 2015-08-10:
still as of IMG 138

jriddell, 2015-09-24:
it's because we pin to prefer the mobile kci qt package but there's a newer version in launchpad which is installed on the image.  does it cause any problems?

bhush9, 2015-11-26:
No longer happens now as we are using qt 5.5 from our CI.

closed 2015-11-26 by star-buck