blue-systems/plamo-initial-dev_DEPRECATED #173: [packages]: move everything from ppa -> mobile.kci.pangea.pub

star-buck, 2015-07-31:
So the idea was to no longer need to build stuff on LP:
https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+packages

Can we therefore
* delete all packages already build in our own kci repo and 
* move the remaining packages from this ppa to mobile.kci.pangea.pub

Or is there a reason that somehow requires them to still be in the ppa?

apachelogger, 2015-09-24:
Needs technology to build arbitrary things via jenkins first. Which has been on todo for like 5 months -.-

bhush9, 2015-11-21:
I think, now we can safely remove at least this from this ppa:

- samegame : packaged on MobileKCI already
- consolekit(2) : I believe it was added to fix polkit issue mentioned in #130 but no longer required
- libepoxy : already in mobile CI
- voicecall and friends (libresource, libresourceqt) : voicecall dependency was removed from the plasma-phone-components quite sometime ago.
- qtwayland : we already build qt 5.5 on mobile CI

I don't have power to dump things..

bhush9, 2015-11-21:
Things I am not sure about are:

- ubuntu-touch-session : why it is pinned to specific version?
- signon-ui : do we ship some specific patches?
- telepathy-mission-control : same question as signon-ui (maybe we can add them in CI?)
- libngf-qt : ????!
- dekko : ????!

mck182, 2015-12-01:
signon-ui - yes, requires a special patch for vivid [1]
telepathy-mission-control - I believe that this also still requires a patch for vivid [2]

[1] - https://code.launchpad.net/~mardy/signon-ui/lp1473422
[2] - https://bugs.freedesktop.org/show_bug.cgi?id=91272 and https://bugs.launchpad.net/ubuntu/+source/telepathy-mission-control-5/+bug/1473011

star-buck, 2015-12-16:
URL is no longer kci, but neon: http://mobile.neon.pangea.pub/

star-buck, 2016-03-02:
@bhush9 : is this done or are there still things added to our IMGs via PPAs that we control?

bhush9, 2016-03-02:
Unfortunately yes, still we need ppa, at least for ubuntu-touch-session and telepathy-mission-control-5

star-buck, 2016-03-02:
cant we built the packages and put them into our own repo?
http://mobile.neon.pangea.pub/

bhush9, 2016-03-02:
We need some patches on top of them, which means forking whole source package and later keeping them in sync when things change upstream. I would like to avoid them.. our CI have tech to build things as it is... but not with small patches.

bhush9, 2016-07-27:
So... I've migrated away from the ubuntu-touch-session package, 20160727-062941 doesn't use PPA at all, previously 20160727-062941 attempted to do same, but failed because ubuntu-touch-session have extrausers pam config, which I've copied to plasma-phone-settings package now...

bhush9, 2016-07-27:
I will proceed to delete PPA now, as it have all critical use-cases covered.

star-buck, 2016-07-27:
ill close this here then

closed 2016-07-27 by star-buck