blue-systems/plamo-initial-dev_DEPRECATED #177: [Packaging] KConfig cmake files

aleixpol, 2015-08-07:
Hi,
The package `libkf5config-bin-dev` should include the files:
```
/usr/lib/arm-linux-gnueabihf/cmake/KF5Config/KF5ConfigCompilerTargets.cmake
/usr/lib/arm-linux-gnueabihf/cmake/KF5Config/KF5ConfigCompilerTargets-debian.cmake
```

This is because we need to be able to have `libkf5config-bin-dev:amd64` installed together with `libkf5config-dev:armhf`.

jriddell, 2015-09-17:
<sitter> not sure that makes sense
<sitter> libkf5config-dev:armhf would contain /usr/lib/arm-linux-gnueabihf/cmake/KF5Config
<sitter> libkf5config-bin-dev:amd64 would contain /usr/lib/amd64-linux-gnu/cmake/KF5Config
<sitter> an rdep would find former and former would somehow have to get latter (which literally makes no sense as then it might as well not find latter and simply find the binaries :P)
<sitter> also libkf5config-dev:amd64 and libkf5config-bin-dev:amd64 would conflict then
<sitter> I feel like aleix and I discussed this at some point and devised a cmake level solution

bhush9, 2016-07-26:
I am not sure what is to be done here? or anything et all?

star-buck, 2016-07-26:
@aleixpol : this be closed?

closed 2017-12-09 by star-buck