blue-systems/plamo-initial-dev_DEPRECATED #181: browser: failing to start

star-buck, 2015-08-10:


star-buck, 2015-08-10:
failing to start since IMG 138

bhush9, 2015-08-11:
webbrowser-app: eglplatformcommon.cpp:256: const char* eglplatformcommon_eglQueryString(EGLDisplay, EGLint, const char* (*)(EGLDisplay, EGLint)): Assertion `ret != __null' failed.

It asserts this way...

bhush9, 2015-08-11:
Also,

[0903/135856:WARNING:oxide_qt_gl_context_dependent.cc(76)] Unable to create adopted GL context for platform: wayland - unrecognized platform
[0903/135856:WARNING:oxide_qt_browser_startup.cc(161)] Cannot determine GL implementation to use - unrecognized Qt platform: wayland

Not sure whatever changed between two images

bhush9, 2015-08-24:
Should we switch browser to anglefish instead of ubuntu's web browser application?

sebasje, 2015-08-31:
Angelfish doesn't build on ARM (due to webengine not building on ARM either), so that's not an option for now. :/

mgraesslin, 2015-08-31:
maybe we can use https://github.com/Dax89/harbour-webpirate - it's QtWebKit based but works rather nicely on the Jolla (considerably better than the default gecko based browser)

llelectronics, 2015-08-31:
And I am the developer of this qtwebkit based Sailfish browser: https://github.com/llelectronics/webcat
Though I would not recommend to use any for the long run as qtwebkit is a pain in the b*tt. To many workarounds are necessary for it to run good.

mgraesslin, 2015-08-31:
just installed webcat, will give it a try :-)

bhush9, 2015-10-13:
So.. A quite big update:

My hunt for the proper™ web browser ended with good old Mozilla firefox. I learned that firefox have armhf builds available in ubuntu repos. I installed it to give it try.

https://launchpad.net/ubuntu/vivid/armhf/firefox

One of the major problem was keyboard. Firefox being GTK(2?) application was not playing right with maliit. It required packaging maliit-inputcontext-gtk https://github.com/maliit/inputcontext-gtk and some other config in place.

Other problem is HighDPI. but it is easily solved by adjusting layout.css.devPixelsPerPx from about:config.

Inserting mandatory screenshot.

[image]

llelectronics, 2015-10-13:
There is qtmozembed which is a port of the gecko (firefox) engine to qt5 + qml: https://github.com/tmeshkova/qtmozembed
It is used by SailfishOS for their default browser which is also opensource: https://github.com/sailfishos/sailfish-browser
Only complain I have about it it uses tons of memory in comparison to QtWebkit. (but is much more compatible regarding html5)
Best would be anything Oxide or QtWebEngine based in my opinion still.

bhush9, 2015-10-13:
I tried to build qtmozembed but mostly it is dependency mess.. :(

mgraesslin, 2015-10-18:
Concerning the problem of insufficient touch support in Firefox: I noticed the other day that Xwayland (at least master) has now proper touch support. So rolling out a newer X stack might improve the situation (let's wait for upgrade to Kubuntu 5.10 first).

notmart, 2015-10-19:
we would have to undo our wrapping to mouseevents tough?

mgraesslin, 2015-10-19:
> we would have to undo our wrapping to mouseevents tough? what adaptions kwayland would need to have?

no, that can and should stay. We were smart and only enabled it if the application does not support touch.

star-buck, 2015-11-15:
closing this here.

closed 2015-11-15 by star-buck