blue-systems/plamo-initial-dev_DEPRECATED #186: [homescreen]: make horizontal layout propotional

star-buck, 2015-08-12:
if you run Plasma Mobile shell in horizontal mode on a wetab, laptop or desktop, make the top panel, iconsizes and bottom panel propotional sizes, so it's nicely useable (to try use KCI Mobile http://mobile.kci.pangea.pub/iso/kubuntu-phone_201508111051-amd64.iso).

This would also be a good pre-requisite for devices which are currently used in standard vertical mode and could be rotated at a later point.

bhush9, 2015-09-19:
I think this is done mostly : You can test it in latest iso http://arch.pangea.pub/ISO/plasma-mobile/plasma-arch-20150918-x86_64.iso

Fixed issues :

- Icon sizes for launchers
- Panel sizes
- Font sizes (ArchISO follows defaults whatever is on desktop) and can be configured via Appearance KCM

star-buck, 2015-09-24:
works on arch
lets wait for closing until seen working on an actual kubuntu mobile ISO

bhush9, 2015-09-30:
http://arch.pangea.pub/ISO/plasma-mobile/kubuntu-phone_201509301520-amd64.iso

418eca2739ef6d804f2e8954498c137d  kubuntu-phone_201509301520-amd64.iso

(Note this ISO have just updates of existing packages, any additional packages are not included due to CI being broken)

star-buck, 2015-10-03:
works, closing.

closed 2015-10-03 by star-buck