blue-systems/plamo-initial-dev_DEPRECATED #188: plasmashell doesn't restart itself on crash

bhush9, 2015-08-25:
It restarts but gets stuck

kscreen: launcherDataAvailable: "org.kde.KScreen.Backend.QScreen"
kscreen: Launcher finished with exit code 1 , status 0
kscreen: Service for requested backend already running
QProcess: Destroyed while process ("/usr/lib/arm-linux-gnueabihf/libexec/kf5/kscreen_backend_launcher") is still running.
kscreen: Launcher finished with exit code 0 , status 1
Error found while setting up ShellCorona's KScreen:  "Message did not receive a reply (timeout by message bus)"

and I can not get good backtrace from gdb..

bhush9, 2015-09-14:
I did experiment little bit more after @notmart mentioned this on IRC

So my theory is

* Plasmashell crashes and child kscreen backend launcher starts to clean up
* New plasmashell re-spawns and thinks kscreen backend is running and doesn't start new instance

```
kscreen: Service for requested backend already running
QProcess: Destroyed while process ("/usr/lib/arm-linux-gnueabihf/libexec/kf5/kscreen_backend_launcher") is still running.
```

* But by the time it actually fetches config process previous process exits safely/

```
kscreen: Launcher finished with exit code 0 , status 1
```

* Now in plasmashell even if this operation have error it just continues by printing it and doesn't do anything as backup plan and then plasmashell doesn't show anything

```
    connect(new KScreen::GetConfigOperation(KScreen::GetConfigOperation::NoEDID), &KScreen::GetConfigOperation::finished,
                this, [this](KScreen::ConfigOperation *op) {
                    if (op->hasError()) {
                        qWarning() << "Error found while setting up ShellCorona's KScreen: " << op->errorString();
                    }
                    m_screenConfiguration = qobject_cast<KScreen::GetConfigOperation*>(op)->config();
                    load();
                });
```

I am not sure why we just carry on when ```op->hasError()``` and just print warning.. @aleixpol any hints? this if was added by you in https://quickgit.kde.org/?p=plasma-workspace.git&a=commit&h=cd946735793ad

```
Error found while setting up ShellCorona's KScreen: "Message did not receive a reply (timeout by message bus)"
```

Possible solution is to don't just continue when operation has error. Try again or perhaps just abort. 

As a workaround for now I will write patch to start kscreen_backend_launcher from the plasma-phone script before starting plasmashell, that way it doesn't get killed when plasmashell aborts.

bhush9, 2015-09-15:
Pushed workaround in : http://commits.kde.org/plasma-phone-components/bbf9c0c

Will try to solve it properly

jriddell, 2015-09-16:
looks built

bhush9, 2015-09-16:
Erm actually lets keep this open.. its workaround-ed and not properly fixed.

bhush9, 2015-10-10:
Fixed properly with commit in libkscreen..

In IMG9 20151010.081111

closed 2015-11-15 by star-buck