blue-systems/plamo-initial-dev_DEPRECATED #189: [Qt] integrate patches for inline edit menus

notmart, 2015-09-01:
Qt (qtquickcontrols) needs the two following (already upstreamed, but only in dev, so 5.6) patches:
https://codereview.qt-project.org/#/c/121709/
https://codereview.qt-project.org/#/c/121659/

this will make the plasma feature in
https://git.reviewboard.kde.org/r/124374/
actually work

notmart, 2015-09-07:
another patch needed:
https://codereview.qt-project.org/#/c/125032/

jriddell, 2015-09-08:
https://codereview.qt-project.org/#/c/121709/ doesn't apply to the version of Qt we're using, 5.4.1.  Could you remake it for this version?

notmart, 2015-09-09:
this is the patch redone on 5.4
https://paste.kde.org/p8tmyfrbn

jriddell, 2015-09-14:
Uploaded qtquickcontrols-opensource-src_5.4.1-0ubuntu1plasma1 to mobile kci with your 2 patches

closed 2015-09-16 by jriddell