blue-systems/plamo-initial-dev_DEPRECATED #190: Start testing the Plasma Mobile SDK from mobile KCI

aleixpol, 2015-09-01:
Here's a test script
http://quickgit.kde.org/?p=xbuilder.git&a=blob&f=test%2Fdocker-test.rb&o=plain

I'm unsure how to proceed.

aleixpol, 2015-09-15:
Ping?

notmart, 2015-09-16:
testing it (had to change /bin/ruby to /usr/bin/ruby and had to run the script as root)

what i should look for?

bhush9, 2015-09-16:
"had to run the script as root"

You have to add your user to ```docker``` group

bhush9, 2015-09-26:
I am confused.. you want this on plasma mobile ISO? or want to CI'fy this at mobile.kci.pangea.pub:8080 ??

star-buck, 2015-09-26:
imo start testing the SDK by installing it under a running mobile KCI ISO.
so not install it default on the ISO.

aleixpol, 2015-09-27:
Well, the plan is to have it running on the servers, so that we can get a good-old red light when it stops working. It can stop working mostly for packaging issues (i.e. multiarch) so I'd say it makes sense for it to stay in mobile.kci.pangea.pub:8080 and it was implemented as Harald suggested to that end, providing a script that would be supposedly trivial to integrate in that jenkins instance.

jriddell, 2015-10-02:
Add to jenkins, sorry for the delay
http://mobile.kci.pangea.pub:8080/job/xbuilder

aleixpol, 2015-10-02:
Looks good, thanks a lot!!

closed 2015-10-02 by aleixpol

bhush9, 2015-10-14:
mmm.. this script needs to be changed in a way so that it doesn't pickup cached image and just move on..

http://mobile.kci.pangea.pub:8080/job/xbuilder/

this script is not actually building image and installing packages but just picking things from cache and moves on..