blue-systems/plamo-initial-dev_DEPRECATED #191: [IMG]: "(r/d)ebrand" IMG to "kde neon" mobile

star-buck, 2015-09-08:


jriddell, 2015-09-14:
I've pushed the code to kde:mobster
rebranding will have to wait for e.v. to make up its mind in a ten days

star-buck, 2015-09-14:
Any link?
Do we have a phabricator project for it?

jriddell, 2015-09-15:
https://projects.kde.org/projects/playground/devtools/mobster
No phabricator for it yet, there's already bugs for it reported here so I wasn't wanting to add another bug tracker

jriddell, 2015-09-24:
done as much as is sane for now

star-buck, 2015-11-15:
Update Note:

Since we now have 
http://neon.pangea.pub
I following old kci logic and added 
http://mobile.neon.pangea.pub

star-buck, 2015-11-15:
For now mobile.neon points to same server as mobile.kci.

bhush9, 2015-11-15:
Is neon.pangea.pub down or doesn't point to anything yet? it says ERR_CONNECTION_REFUSED for me?

star-buck, 2015-11-15:
neon.pangea.pub may not be setup completely serverside yet. but both domain and DO server exist.

star-buck, 2015-12-27:
I also updated the flashing instructions here to read neon instead of kubuntu:
http://neon.plasma-phone.org/

Lets make sure it stays consistent with the next phone IMG update (keeping a screenshot saved for reference below):
[image]

star-buck, 2015-12-27:
maybe replace string "ubuntu" in 
"Latest available Image: ubuntu=2015...." 
also with "neon"?

bhush9, 2015-12-28:
We can change it in website, but internally it will still be called ubuntu=.. as this variable is used by ubuntu-device-flash internally.

star-buck, 2015-12-28:
Okay, thats fine internally, so if it doesnt interfere with that, lets
change it visibly on website only.

star-buck, 2016-03-02:
closing this for now

closed 2016-03-02 by star-buck