blue-systems/plamo-initial-dev_DEPRECATED #192: Package Marble Maps

notmart, 2015-09-09:
package marble maps, from the marble repository, master branch.
at the moment marble maps doesn't have a desktop file, so may have to be delayed, but they promised one.
also, the package should be splitted to have marble desktop and marble maps in different packages

aleixpol, 2015-09-09:
Having heard the conversation, I think it would be useful to provide them a Review Request with a patch that adds marble-mobile.desktop file...

jriddell, 2015-09-15:
I've updated the packaging and it's building now in kci mobile

jriddell, 2015-09-15:
built now on armhf, waiting on publisher to stop being busy to publish

jriddell, 2015-09-16:
built

star-buck, 2015-09-16:
reopen - not closing, but mark "Ready for QA" until confirmed fixed on an actual IMG.

star-buck, 2015-09-24:
not seeing any marble in v4 IMG...

apachelogger, 2015-09-24:
@jriddell maybe not installed by mobster?

jriddell, 2015-09-24:
coming in today's new image

jriddell, 2015-09-24:
on image 20150924.130033

star-buck, 2015-09-24:
marble maps icon on homescreen, but app doesnt start

star-buck, 2015-09-24:
closing this now and open new ticket

closed 2015-09-24 by star-buck