blue-systems/plamo-initial-dev_DEPRECATED #194: I want images for the E5

jriddell, 2015-09-16:
I want images for the E5 says rohan

shadeslayer, 2015-09-16:
Would be cool to have, I'm thinking of going to ubucon.de , and it'd be nice to show it off there.

jriddell, 2015-09-16:
<Riddell> shadeslayer: ta da http://kubuntu.plasma-mobile.org/kubuntu-phone/devel-proposed-e5/vegetahd/index.json
<Riddell> shadeslayer: mirrored at http://mobile.edinburghlinux.co.uk/ if you want faster download
<Riddell> shadeslayer: I have zero idea if that'll work, it's just taking the vivid-proposed ubuntu-touch device image for vegetahd and putting our image onto it

shadeslayer, 2015-09-16:
Doesn't work, where be the logs from flashing?

star-buck, 2015-11-15:
lets try to get things working again properly on one device the nexus5, closing this for now.

closed 2015-11-15 by star-buck

bhush9, 2015-12-08:
I am removing the channel for this from kubuntu.plasma-mobile.org, it somehow makes publishing of other images fail (normal Nexus5 images)

```
2015-12-08 08:51:17,864 INFO New file from 'version': neon-mobile/devel-proposed-e5/vegetahd/version-12.tar.xz
2015-12-08 08:51:17,864 INFO Generating delta from '11' for 'version'
2015-12-08 08:51:17,865 INFO Publishing new image '12' (ubuntu=20151126.052251,keyring=archive-master,version=12) with 4 files.
Traceback (most recent call last):
  File "./bin/import-images", line 276, in <module>
    version_detail=",".join(environment['version_detail']))
  File "/home/ubuntu/kubuntu/system-image/bin/../lib/systemimage/tree.py", line 831, in create_image
    "%s.asc" % abspath)
Exception: The GPG file signature doesn't exists: /home/ubuntu/kubuntu/system-image/www/pool/device-11ab2cbf36949374b8b43f02283a6dfd1a147dc1d131c3629ec54700bd6a477c.tar.xz.asc
Traceback (most recent call last):
  File "./proposed-is-good", line 44, in <module>
    os.symlink(workingDirectory + "mobster/"+linkFile, linkFile)
OSError: [Errno 17] File exists
```