blue-systems/plamo-initial-dev_DEPRECATED #195: [Settings]: add "About" kcm

star-buck, 2015-09-17:
so people see info about Distro name, Qt, Plasma versioning

jriddell, 2015-09-17:
also needs a neon-settings package added

jriddell, 2015-09-24:
about kcm is part of kinfocenter

star-buck, 2015-09-24:
right, since its kci mobile, would refering to it as "neon"(-mobile) too, like kci desktop ISOs are labelled neon-files:
http://files.kde.org/snapshots/neon5-latest.iso.mirrorlist

star-buck, 2015-12-16:
img13: still shows "Kubuntu" + logo in kinfocenter

star-buck, 2016-03-02:
@bhush9 :
would be great if you look how to get it re-named and a more neutral logo (like "plasma" dots) shown, so once we also offer other builds from e.g. suse or debian, we know how to do this.

jriddell, 2016-09-20:
neon-settings has the required files for this
>cat /etc/xdg/kcm-about-distrorc
[General]
LogoPath=/usr/share/pixmaps/neon.png
Version=Developer Edition
Variant=Unstable Branches

I guess it needs the equivalent added to mobile's settings package

star-buck, 2018-02-08:
@bhush9 : kinfocenter now missing on current img?

star-buck, 2018-02-08:
closing here, just needs kinfocenter installed again then works fine

closed 2018-02-08 by star-buck