blue-systems/plamo-initial-dev_DEPRECATED #197: [camera]: worked before, now only shows dark background

star-buck, 2015-09-24:


bhush9, 2015-09-25:
Same gstreamer mess like issue #201 

I think we need to build qt5 with gstreamer 1.0

bhush9, 2015-11-28:
Now one no-longer gets gstreamer errors like that but now instead rainbow bars are shown... meh..

star-buck, 2015-12-16:
img13: now doesnt even start anymore

bhush9, 2015-12-17:
Yeah, camera is known to break.. our newer libhybris is incompatible with qtubuntu-camera plugin for qtmultimedia. I am working on solution.. most probably it will be fixed in next image or image+1.

star-buck, 2016-03-02:
camera app shows colored standstill image, but doesnt seem to take pictures.

bhush9, 2016-07-27:
I at one point started qtmultimedia plugin for the camera which used android HAL api's.. however I didn't make much progress, mainly because I don't fully understand android API etc. I've one possible solution which is compiling compat module of libhybris with CM image.. and use qtubuntu-camera from ubuntu.

Though we wouldn't want to compile in compat module as it would conflict with input in kwin_wayland.. Leaving no choice but broken camera for now.

bhush9, 2017-12-14:
This will need the gst-droid packaged. I will include it in next image, I already have it working locally.

bhush9, 2017-12-19:
Update: Everything needed is packaged, but needs to be bundled inside the image. However there is a conflict between the codecs installed in ticket #277 as mentioned there. There is option to solve it which is gst-omx, and I've it packaged also, most of the formats are also working there with hardware accelerated video.. :-)

However it is possible that my testing is limited, so i haven't shipped it in image yet, @star-buck can you tell me which videos you have tested, so I can test it with newer setup and fix where fixes are needed?

So far in my testing following works:

- Big buck bunny 1080P, mp4, 30 and also 60 FPS
- Sintel trailer, mkv, 1080P (? FPS)
- Youtube: Star wars, Last jedi trailer (HD)
- Some more random youtube trailers like daredevil, iron-man etc..

And following doesn't work:

- Tears of steel 1080P, WebM, 60 FPS (it loads , but drops frames as movie progresses, however 1080P, MP4 works)

bhush9, 2017-12-22:
@star-buck is camera required for demo at Dec 27? If so I can roll a image with camera fix..

star-buck, 2017-12-22:
Nice to have yes, but not really needed.

star-buck, 2018-02-08:
@bhush9 : please ship fix.

bhush9, 2018-02-09:
`20180209-155413` fixes this