blue-systems/plamo-initial-dev_DEPRECATED #201: [marble maps]: icon in homescreen, but app doesnt start

star-buck, 2015-09-24:


bhush9, 2015-09-25:
Its happening due to mixing up gst 0.10 and gst 1.0

Perhaps marble needs to be compiled against gstreamer 0.10

notmart, 2015-09-28:
why both versions are installed?
this would suggest that our qt is compiled funny since it happens to multiple applications like the camera?

bhush9, 2015-09-28:
Having both versions installed is not really problem IIRC.. but linking to both in same binary is problem.

notmart, 2015-09-28:
so the problem is how the particular application is built by cmake?

bhush9, 2015-09-28:
Yeah.. 

11:20:09 Get: 178 http://ports.ubuntu.com/ubuntu-ports/ vivid/main libgstreamer1.0-0 armhf 1.4.5-1 [612 kB]
11:20:10 Get: 179 http://ports.ubuntu.com/ubuntu-ports/ vivid/main liborc-0.4-0 armhf 1:0.4.22-1 [121 kB]
11:20:10 Get: 180 http://ports.ubuntu.com/ubuntu-ports/ vivid/main libgstreamer-plugins-base1.0-0 armhf 1.4.5-1ubuntu1 [439 kB]

for marble.. and this is case only for marble.. I am not sure what pulls gstreamer1.0 for marble..

On other hand this doesn't explain why camera application is not working with same problem.. Perhaps qt is also built with two gstreamer..

:fearful:

notmart, 2015-09-28:
the camera application (do both of them, ubuntu and ours have the same problem?) are just using the qtmultimedia imports, so yeah, the problem is in the Qtmultimedia build i guess

bhush9, 2015-09-28:
Not sure,

I can not install it

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Starting pkgProblemResolver with broken count: 1
Starting 2 pkgProblemResolver with broken count: 1
Investigating (0) qtdeclarative5-ubuntu-thumbnailer0.1 [ armhf ] < none -> 2.3+15.04.20150915.1-0ubuntu1 > ( oldlibs )
Broken qtdeclarative5-ubuntu-thumbnailer0.1:armhf Depends on qml-module-ubuntu-thumbnailer0.1 [ armhf ] < none -> 2.3+15.04.20150915.1-0ubuntu1 > ( libdevel )
  Considering qml-module-ubuntu-thumbnailer0.1:armhf 1 as a solution to qtdeclarative5-ubuntu-thumbnailer0.1:armhf 0
  Holding Back qtdeclarative5-ubuntu-thumbnailer0.1:armhf rather than change qml-module-ubuntu-thumbnailer0.1:armhf
Investigating (1) camera-app [ armhf ] < none -> 3.0.0+15.04.20150707-0ubuntu1 > ( x11 )
Broken camera-app:armhf Depends on qtdeclarative5-ubuntu-thumbnailer0.1 [ armhf ] < none -> 2.3+15.04.20150915.1-0ubuntu1 > ( oldlibs )
  Considering qtdeclarative5-ubuntu-thumbnailer0.1:armhf 0 as a solution to camera-app:armhf 9999
    Reinst Failed early because of libqt5quick5:armhf
    Reinst Failed early because of libqt5quick5-gles:armhf
    Reinst Failed because of qml-module-ubuntu-thumbnailer0.1:armhf
  Considering qtdeclarative5-ubuntu-thumbnailer0.1:armhf 0 as a solution to camera-app:armhf 9999
Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 camera-app : Depends: qtdeclarative5-ubuntu-thumbnailer0.1 but it is not going to be installed
E: Unable to correct problems, you have held broken packages

bhush9, 2015-09-28:
Long term solution

Lets upgrade to 5.5 and build it with gstreamer 1.0

https://wiki.qt.io/New_Features_in_Qt_5.5

GStreamer 1.0 support. Note that the default is still 0.10. 1.0 support can be enabled by configuring Qt with '-gstreamer 1.0'.

star-buck, 2015-10-10:
note: neon-img v9: stll doesnt come up

star-buck, 2015-11-15:
img11: still not working

bhush9, 2015-11-26:
Apparently now we have qt5.5 and still it doesn't start due to some symbol mismatch.. needs investigation.

bhush9, 2015-11-28:
Starts with IMG12

star-buck, 2015-12-16:
it doesnt, and also doesnt with img13

bhush9, 2015-12-17:
fun.. with IMG12 it was starting and working and (IMG12 was promoted to devel channel to IMG7) but not in IMG13.. dunno what changed in between.

bhush9, 2015-12-17:
Rebuilding marble in hope it fixes it.

bhush9, 2015-12-18:
yep, rebuild seems to have solved this..

star-buck, 2015-12-18:
still doesnt start... how do you test that it starts for you?
with nexus 5, it didnt start in img12, 13, 14...

bhush9, 2015-12-18:
oops.. indeed it is broken. so well after rebuild I upgraded it in previous image and that seemed to work.. but now in image it is broken.. something seems to be wrong with packaging.. I will investigate..

bhush9, 2015-12-18:
Right, @star-buck if you upgrade, marble-maps will be working.. there are some stray packages in repos that gets picked while creating IMG so it doesn't work in raw image. but after upgrade marble starts again. I will remove stray packages tomorrow.

star-buck, 2015-12-18:
how do i update? it says i got the latest version installed...

bhush9, 2015-12-19:
Not marble but if you update all packages, I.e sudo apt-get update && sudo apt-get upgrade

star-buck, 2015-12-19:
Yeah! it starts.

closed 2015-12-19 by star-buck