blue-systems/plamo-initial-dev_DEPRECATED #204: Switch to qt5.5 for mobile images

bhush9, 2015-09-28:
Multiple reasons -

* We want to increase required dep for plasma-workspace/desktop
* Qt 5.5 brings various fixes for instance, gstreamer 1.0 support in Qtmultimedia which would fix #201 as well as #197

jriddell, 2015-09-28:
I want to switch to wily, images are available for wily for nexus 5 it seems.

Harald what's needed to get wily archives in mobile kci?

Note that Wily is still Qt 5.4.2 but at least there are PPAs to get 5.5 packages from

apachelogger, 2015-09-28:
@jriddell you can't right now because docker management is still broken. but once that is fixed...

To switch off the top of my head:
- Repo: Change pangea-kitchen databag of repo-mci to accept wily and knife solo cook taspar
- Jenkins: in jenkins_jobs_update search for vivid and replace with wily. run script. should be everything.
- ci-tooling: in kci/binarier.rb switch vivid for wily

For jenkins you could also write a script to rename all jobs to s/vivid/wily and then run the updater. That would preserve the history but quite frankly I doubt it would be useful.
If you choose to simply create new jobs for wily (i.e. don't rename the vivid jobs) don't forget to use jenkins_delete.rb to delete the vivid jobs.

I'll do the repo thing right now though, it has no adverse effects on vivid.

jriddell, 2015-09-30:
Added wily jobs but on building them it complains
http://mobile.kci.pangea.pub:8080/job/wily_unstable_threadweaver_src/2/console
12:52:18 /var/lib/jenkins/.gem/ruby/2.2.0/gems/docker-api-1.21.4/lib/docker/connection.rb:46:in `rescue in request': No such image: jenkins/wily_unstable (tag: latest) 
how to fix that?

apachelogger, 2015-09-30:
Needs http://mobile.kci.pangea.pub:8080/job/mgmt_docker_vivid/ copied and changed for wily

jriddell, 2015-09-30:
got jobs added but builders need an apt update somewhere
 http://mobile.kci.pangea.pub:8080/job/mgmt_docker_wily/label=46.101.251.91/2/console 
<sitter> Riddell: really I'd just postpone this until after the bloody tooling deployment is working again

bhush9, 2015-11-26:
Qt 5.5 now available in IMG12

closed 2015-11-26 by star-buck