blue-systems/plamo-initial-dev_DEPRECATED #207: [package] Latest libhybris

mgraesslin, 2015-10-08:
Hi packagers,

in order to address the flickering on the bottom panel I need a new libhybris. Please try to package the latest master. DO NOT INCLUDE IT IN ANY REPO! Libhybris master is API/ABI incompatible to the one we have. If we switch to it, we need to make sure that KWin is adjusted at that time.

So please provide the packages somewhere where I can easily add it to the phone and dev system so that I can try to adjust KWin and get everything adjusted to it

jriddell, 2015-10-08:
Not an easy thing to do alas.
libhybris we have is from ubuntu who use a 2013 snapshot with a large patch and compiles with a equally old version of package called android-headers
github.com/libhybris needs a newer android-headers to bring in android-config.h but I've no idea where that comes from and of course the ubuntu patch doesn't apply
suggestions welcome

bhush9, 2015-10-08:
What do wily uses?

mgraesslin, 2015-10-08:
@jriddell thanks for starting looking into it. I feared that it will be complex

> suggestions welcome

AFAIK the android-config.h is just a renamed file. A symlink to /usr/include/android/android-version.h might work.

Please do not try to apply Ubuntu's patch. That's what boud tried in the past and it just didn't work. I want upstream libhybris ;-)

> What do wily uses?
according to http://packages.ubuntu.com/wily/libhybris it's still 2013

jriddell, 2015-10-08:
@bhush9 wily uses much the same as vivid, a snapshot from 2013 with a muckle patch

jriddell, 2015-10-08:
@mgraesslin well impressively it started to compile then got stuck on
hardware_legacy/vibrator.h
I wonder where to get that file

jriddell, 2015-10-08:
well amazingly that seems to have compiled, voila
http://weegie.edinburghlinux.co.uk/~jr/tmp/libhybris/
straight from github.com/libhybris/libhybris without patches

mgraesslin, 2015-10-08:
awesome! Is there a convenient dpkg command to get them all installed at the same time?

@davidedmundson can you provide me your patch for KWin to use more recent libhybris?

jriddell, 2015-10-08:
wget -r http://weegie.edinburghlinux.co.uk/~jr/tmp/libhybris/; sudo dpkg --install weegie.edinburghlinux.co.uk/~jr/tmp/libhybris/*deb

well it might work

mgraesslin, 2015-10-09:
@jriddell the packages indicate gcc5 - can that cause problems given that KWin is compiled with gcc 4.9?

mgraesslin, 2015-10-09:
> well it might work

no it doesn't ;-)

sudo dpkg -i weegie.edinburghlinux.co.uk/~jr/tmp/libhybris/libhybris-dev_0.1.0+git20131207+e452e83-0ubuntu40~gcc5.1_armhf.deb
(Reading database ... 88577 files and directories currently installed.)
Preparing to unpack .../libhybris-dev_0.1.0+git20131207+e452e83-0ubuntu40~gcc5.1_armhf.deb ...
Unpacking libhybris-dev (0.1.0+git20131207+e452e83-0ubuntu40~gcc5.1) over (0.1.0+git20131207+e452e83-0ubuntu39) ...
dpkg: error processing archive weegie.edinburghlinux.co.uk/~jr/tmp/libhybris/libhybris-dev_0.1.0+git20131207+e452e83-0ubuntu40~gcc5.1_armhf.deb (--install):
 trying to overwrite '/usr/include/KHR/khrplatform.h', which is also in package libegl1-mesa-dev:armhf 10.5.9-2ubuntu1~vivid2
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 weegie.edinburghlinux.co.uk/~jr/tmp/libhybris/libhybris-dev_0.1.0+git20131207+e452e83-0ubuntu40~gcc5.1_armhf.deb

mgraesslin, 2015-10-09:
Next problem: the package libhybris-dev no longer contains /usr/lib/arm-linux-gnueabihf/libis.so (and a few other sos are missing)

mgraesslin, 2015-10-09:
good news: KWin starts and renders with the new libhybris. But input is not working due to the missing libis.so. Very promising result!

mgraesslin, 2015-10-09:
But GL apps make KWin crash in libhybris:
Program received signal SIGSEGV, Segmentation fault.
0xb528096e in eglDestroyImageKHR ()
   from /usr/lib/arm-linux-gnueabihf/libhybris-egl/libEGL.so.1
(gdb) bt
#0  0xb528096e in eglDestroyImageKHR ()
   from /usr/lib/arm-linux-gnueabihf/libhybris-egl/libEGL.so.1
#1  0xb6f2d84c in KWin::AbstractEglTexture::~AbstractEglTexture() ()
   from /usr/lib/arm-linux-gnueabihf/libkwin.so.5
#2  0xb2228cd6 in KWin::EglHwcomposerTexture::~EglHwcomposerTexture() ()
   from /usr/lib/arm-linux-gnueabihf/qt5/plugins/org.kde.kwin.waylandbackends/KWinWaylandHwcomposerBackend.so
#3  0xb2228cfe in KWin::EglHwcomposerTexture::~EglHwcomposerTexture() ()
   from /usr/lib/arm-linux-gnueabihf/qt5/plugins/org.kde.kwin.waylandbackends/KWinWaylandHwcomposerBackend.so
#4  0xb6ec3c74 in QExplicitlySharedDataPointer<KWin::GLTexturePrivate>::operator=(KWin::GLTexturePrivate*) () from /usr/lib/arm-linux-gnueabihf/libkwin.so.5
#5  0xb6ebade2 in KWin::SceneOpenGL::Texture::load(KWin::WindowPixmap*) ()
   from /usr/lib/arm-linux-gnueabihf/libkwin.so.5
#6  0xb6ebca46 in KWin::OpenGLWindowPixmap::bind() ()
   from /usr/lib/arm-linux-gnueabihf/libkwin.so.5
#7  0xb6ebaff0 in KWin::SceneOpenGL::Window::bindTexture() ()
   from /usr/lib/arm-linux-gnueabihf/libkwin.so.5
#8  0xb6ebb608 in KWin::SceneOpenGL::Window::beginRenderWindow(int, QRegion const&, KWin::WindowPaintData&) () from /usr/lib/arm-linux-gnueabihf/libkwin.so.5
#9  0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

mgraesslin, 2015-10-09:
I hacked around the problem in previous comment, but each GL app results in:
Starting program: /usr/bin/qmlscene test.qml
process 13669 is executing new program: /usr/lib/arm-linux-gnueabihf/qt5/bin/qmlscene
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Created Inout context: MInputContext(0x23aa0)
Using Wayland-EGL
*** Error in `/usr/lib/arm-linux-gnueabihf/qt5/bin/qmlscene': free(): invalid next size (fast): 0x001059b0 ***
[New Thread 0xb39ae410 (LWP 13672)]

Program received signal SIGABRT, Aborted.
0xb5e6e9a6 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
(gdb) bt
#0  0xb5e6e9a6 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#1  0xb5e7c62e in raise () from /lib/arm-linux-gnueabihf/libc.so.6
#2  0xb5e7d332 in abort () from /lib/arm-linux-gnueabihf/libc.so.6
#3  0xb5ea5156 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

mgraesslin, 2015-10-09:
The KWin crash is indeed a bug in KWin now fixed with http://commits.kde.org/kwin/68b7274c5da2c555c7afdaabcec2d84c5e6073e8

Unfortunately GL applications still crash

mgraesslin, 2015-10-09:
Concerning the input (libis.so) problem. That's the legacy stack which got removed from libhybris https://github.com/libhybris/libhybris/commit/a28a3e566e8605d46295967399fffefbb8b4aaaa

Unfortunately we cannot use the new stack. For that we would need hardware/input.h in the android headers.

So either we revert the change or we also need a newer android-headers package.

mgraesslin, 2015-10-09:
The crash of applications is solved. Problem is that I had libegl1-mesa-dev installed which provides libwayland-egl.so which conflicts with the one from libhybris. So this might be a local issue.

mgraesslin, 2015-10-12:
@jriddell can you please repackage with https://github.com/libhybris/libhybris/commit/a28a3e566e8605d46295967399fffefbb8b4aaaa reverted?

jriddell, 2015-10-12:
~ppa2 added to http://weegie.edinburghlinux.co.uk/~jr/tmp/libhybris/ compiled with https://github.com/libhybris/libhybris/commit/a28a3e566e8605d46295967399fffefbb8b4aaaa reverted and the various libraries it took away added back

mgraesslin, 2015-10-12:
awesome, awesome. Everything looks good, review request for KWin is https://git.reviewboard.kde.org/r/125606/

mgraesslin, 2015-10-12:
good news! I got KWin to work with libinput on the phone, so we don't need the patch. See https://git.reviewboard.kde.org/r/125608/

jriddell, 2015-10-13:
new version of libhybris up now on mobile kci for armhf only

mgraesslin, 2015-10-14:
and pushed: http://commits.kde.org/kwin/28fe78a94f7d833eb2624300c409aca0d08cc3c8

Thanks for the help!

closed 2015-10-14 by mgraesslin