blue-systems/plamo-initial-dev_DEPRECATED #209: KAccounts packaging adjustment

mck182, 2015-10-16:
Things got moved around a bit, basically, kaccounts-providers got a new plugin to install, kaccounts-integration lost some plugins and kaccounts-mobile needs packaging. Here's a full list of changes:

Out of kaccounts-integration repo:
src/daemon/plugins/google -> kaccounts-mobile:google/google-contacts-sync
src/daemon/plugins/owncloud -> kaccounts-mobile:common/carddav-contacts-sync
src/owncloud -> kaccounts-providers:plugins/owncloud-ui

Out of kaccounts-providers repo:
services/owncloud-contacts.service.in -> kaccounts-mobile:services/owncloud-contacts.service.in
services/disabled/google-contacts.service.in -> kaccounts-mobile:services/google-contacts.service.in

bhush9, 2015-12-01:
This seems to be done.. please verify everything is fine.. I will add kaccounts-mobile in next image.

bhush9, 2015-12-15:
Available in IMG13.

star-buck, 2016-01-12:
img18 has kaccounts package reported missing under Settings -> Accounts

bhush9, 2016-01-12:
kaccounts-mobile itself is properly packaged, however plasma-settings is missing dep to qml-module-ubuntu-onlineaccounts (installing will fix that, I will change packaging for that)

star-buck, 2016-01-12:
Closing then here.

closed 2016-01-12 by star-buck