blue-systems/plamo-initial-dev_DEPRECATED #213: [kmhajongg]: installs, but doesnt start properly

star-buck, 2015-11-15:


bhush9, 2015-12-18:
from konsole seems to start but being X application it doesn't get input it seems.. (CC: issue #217 )

star-buck, 2015-12-18:
The input is not the problem here, lets make it start correctly as a
launcher first.

star-buck, 2016-02-26:
saw this ticket is double, kmahjongg now works properly when started from console, but still fails to satrt via its own launcher.

bhush9, 2016-07-28:
I just figured this is bug in kmahjongg itself probably.. 

kmahjongg: '<caption>' missing.
kmahjongg: Use --help to get a list of available command line options.

bhush9, 2016-07-28:
Right... so kmahjongg we pull is from ubuntu, and it is kde4 version, we can build on mobile CI if required..

bhush9, 2016-07-28:
http://commits.kde.org/plasma-phone-components/8d31e6b819e147878df01f2f261d8f58108d0754

Fixes this. Mainly this was our launcher removing %c part from the command line. so actual command line,

```kmahjongg -caption %c```

would become,

```kmahjongg -caption```

Which resulted in error provided above, there was implementation to launch application with KRun, but wasn't used actually..

bhush9, 2016-07-28:
I will target this for next week's image.

bhush9, 2016-08-02:
works in last image..

star-buck, 2017-12-09:
needs shipping again, opened another ticket for that

closed 2017-12-09 by star-buck