blue-systems/plamo-initial-dev_DEPRECATED #217: [packaging] Provide XWayland 1.18

mgraesslin, 2015-11-15:
Please provide the latest XWayland server (1.18) in the image. It got released last week and brings important improvements for X11 applications like touch support.

bhush9, 2015-11-15:
Is it available in some ppa or something?

mgraesslin, 2015-11-16:
I do not know but Debian experimental has (working) packages.

star-buck, 2015-12-12:
do we have packages ready for shipping in v13?

bhush9, 2015-12-13:
Unfortunately not yet.. There are no packages available for xwayland 1.18, there are xorg server git ppa named "X crackers" but it doesn't provide xwayland..

star-buck, 2016-01-10:
@bhush9 : please see that this gets packaged now in our own repo and shipped with next img.

bhush9, 2016-01-12:
xwayland | 2:1.18.0-2kubuntu1 | vivid | amd64, armhf

bhush9, 2016-01-12:
IMG18 have it

closed 2016-01-12 by star-buck