blue-systems/plamo-initial-dev_DEPRECATED #218: kscreenlocker packaging

jriddell, 2015-11-18:
<bshah> Riddell: random note: we also want kscreenlocker on mobile CI

shadeslayer, 2015-11-18:
Should just be a matter of running the script that updates the jobs.

bhush9, 2015-11-19:
Requires this against CI tooling

```diff
diff --git a/data/projects_mci.json b/data/projects_mci.json
index 6533fba..150c13b 100644
--- a/data/projects_mci.json
+++ b/data/projects_mci.json
@@ -10,6 +10,7 @@
         "kdeplasma-addons",
         "kfilemetadata",
         "kinfocenter",
+        "kscreenlocker",
         "kwin",
         "kwayland",
         "kwayland-integration",
@@ -126,6 +127,11 @@
             "path": "qt/qtwayland",
             "branch": "kubuntu_vivid_mobile",
             "tarball": "http://http.debian.net/debian/pool/main/q/qtwayland-opensource-src/qtwayland-opensource-src_5.5.1.orig.tar.xz"
+          },
+          {
+            "path": "qt/qtsvg",
+            "branch": "kubuntu_vivid_mobile",
+            "tarball": "http://http.debian.net/debian/pool/main/q/qtsvg-opensource-src/qtsvg-opensource-src_5.5.1.orig.tar.xz"
           }
         ]
       }
```

bhush9, 2015-11-19:
And then upgrade.

bhush9, 2015-11-19:
And also kubuntu_vivid_mobile branch should be created in qtsvg

bhush9, 2015-11-20:
Closing as kscreenlocker build is added in mobile CI.

closed 2015-11-20 by bhush9