blue-systems/plamo-initial-dev_DEPRECATED #219: [packaging] kdeclarative packaging adjustments

bhush9, 2015-11-20:
KDeclarative package doesn't provide /usr/lib/x86_64-linux-gnu/libKF5CalendarEvents.so.5.16.0 which makes things fail with,

```
23:04:55 CMake Error at /usr/lib/x86_64-linux-gnu/cmake/KF5Declarative/KF5DeclarativeTargets.cmake:99 (message):
23:04:55   The imported target "KF5::CalendarEvents" references the file
23:04:55 
23:04:55      "/usr/lib/x86_64-linux-gnu/libKF5CalendarEvents.so.5.16.0"
23:04:55 
23:04:55   but this file does not exist.  Possible reasons include:
23:04:55 
23:04:55   * The file was deleted, renamed, or moved to another location.
23:04:55 
23:04:55   * An install or uninstall procedure did not complete successfully.
23:04:55 
23:04:55   * The installation package was faulty and contained
23:04:55 
23:04:55      "/usr/lib/x86_64-linux-gnu/cmake/KF5Declarative/KF5DeclarativeTargets.cmake"
23:04:55 
23:04:55   but not all the files it references.
23:04:55 
23:04:55 Call Stack (most recent call first):
23:04:55   /usr/lib/x86_64-linux-gnu/cmake/KF5Declarative/KF5DeclarativeConfig.cmake:85 (include)
23:04:55   CMakeLists.txt:41 (find_package)
23:04:55 
23:04:55 
23:04:55 -- Configuring incomplete, errors occurred!
```

jriddell, 2015-11-20:
added but it won't build now because frameworks has been bumped to 5.16 by kubuntu so it needs all of frameworks rebuilt

bhush9, 2015-11-21:
Now, Sounds like other packages like plasma-framework/kcmutils needs to depend upon libkf5calendarevents5 ?

jriddell, 2015-11-21:
discussed on irc, rebuilding

bhush9, 2015-11-21:
Fixed now.

closed 2015-11-21 by bhush9