blue-systems/plamo-initial-dev_DEPRECATED #220: [packaging] Various packaging adjustments required for IMG11

bhush9, 2015-11-22:
- [ ] Marble

Needs depend on qt5script-dev, Fails with

```
03:10:01 CMake Error at CMakeLists.txt:78 (find_package):
03:10:01   By not providing "FindQt5Script.cmake" in CMAKE_MODULE_PATH this project
03:10:01   has asked CMake to find a package configuration file provided by
03:10:01   "Qt5Script", but CMake did not find one.
03:10:01 
03:10:01   Could not find a package configuration file provided by "Qt5Script" with
03:10:01   any of the following names:
03:10:01 
03:10:01     Qt5ScriptConfig.cmake
03:10:01     qt5script-config.cmake
03:10:01 
03:10:01   Add the installation prefix of "Qt5Script" to CMAKE_PREFIX_PATH or set
03:10:01   "Qt5Script_DIR" to a directory containing one of the above files.  If
03:10:01   "Qt5Script" provides a separate development package or SDK, be sure it has
03:10:01   been installed.
03:10:01 
03:10:01 
03:10:01 -- Configuring incomplete, errors occurred!
```

- [ ] breeze

breeze-icons was split out of it, so breeze-icons needs packaging and breeze needs adjustment, amd64 build fails with :

```
02:35:39 dh_install 
02:35:39 cp: 
02:35:39 cannot stat ‘debian/tmp/usr/share/icons/breeze’
02:35:39 : No such file or directory
02:35:39 
02:35:39 dh_install: cp -a debian/tmp/usr/share/icons/breeze debian/breeze-icon-theme//usr/share/icons/ returned exit code 1
02:35:39 debian/rules:24: recipe for target 'override_dh_install' failed
02:35:39 make[2]: *** [override_dh_install] Error 2
```

- [ ] Muon

discover got split out of it needs discover packaged and muon packaging adjusted, fails with

```
03:50:33 dh_install: muon-common missing files (usr/share/icons/hicolor/*/apps/muondiscover.*), aborting
03:50:33 /usr/share/pkg-kde-tools/qt-kde-team/3/dhmk.mk:97: recipe for target 'pre_install_dh_install' failed
03:50:33 make[1]: Leaving directory '/var/lib/jenkins-slave/mci/workspace/vivid_unstable_muon_bin_amd64/muon-5.4.3+git20151122.0343+15.04'
03:50:33 make[1]: *** [pre_install_dh_install] Error 255
03:50:33 make: *** [debian/dhmk_install] Error 2
```

- [ ] Plasma settings

EMAKESNOSENSETOME

```
02:41:35 The following packages have unmet dependencies:
02:41:35  qml-module-qtquick-layouts : Depends: qtbase-abi-5-4-1 which is a virtual package.
02:41:35                               Depends: qtdeclarative-abi-5-4-1 which is a virtual package.
02:41:35  qml-module-qtquick-controls : Depends: qtbase-abi-5-4-1 which is a virtual package.
02:41:35                                Depends: qtdeclarative-abi-5-4-1 which is a virtual package.
02:41:35  qml-module-qtquick-privatewidgets : Depends: qtbase-abi-5-4-1 which is a virtual package.
02:41:35                                      Depends: qtdeclarative-abi-5-4-1 which is a virtual package.
02:41:35  qml-module-qtquick-dialogs : Depends: qtbase-abi-5-4-1 which is a virtual package.
02:41:35                               Depends: qtdeclarative-abi-5-4-1 which is a virtual package.
```

- [ ] plasma-workspace

Requires building qt5sensors on mobile CI

```
04:39:09 The following packages have unmet dependencies:
04:39:09  libqt5sensors5 : Depends: qtbase-abi-5-4-1 which is a virtual package.
```

- [ ] powerdevil

Blocked on plasma-workspace

I think that's it.. there is one more inputcontext-gtk issue but not blocker, and needs that either we fork inputcontext-gtk or merge upstream changes in plasma-maliit-framework.

jriddell, 2015-11-22:
marble does depend on qtscript but I think qtlocation now needs updated to 5.5 to bring in the right version

jriddell, 2015-11-22:
some faff while fixing or working round changes in pangea-tooling (jenkins jobs now have debian repository in name but the old jobs are still around)
qtlocation done
breeze-icons done
discover done
muon compiling away

jriddell, 2015-11-22:
breeze compiling
qtquickcontrols-opensource-src compiling
marble compiling

jriddell, 2015-11-22:
qt sensors compiling
muon done!
breeze done!

bhush9, 2015-11-23:
plasma-workspace building away
plasma-settings building away

marble still failing(?)

bhush9, 2015-11-23:
plasma-workspace and plasma-settings built
powerdevil building

bhush9, 2015-11-26:
marble also fixed.. overall solved.. closing.

closed 2015-11-26 by bhush9