blue-systems/plamo-initial-dev_DEPRECATED #225: [ubuntu apps]: list of apps

star-buck, 2015-11-26:
Browser (https://launchpad.net/ubuntu/+source/oxide-qt)
animal-farm (https://launchpad.net/~plasma-phone/+archive/ubuntu/ubuntu-apps)
Music App (https://launchpad.net/~plasma-phone/+archive/ubuntu/ubuntu-apps)
minesweeper-touch (https://launchpad.net/~plasma-phone/+archive/ubuntu/ubuntu-apps)
ubuntu-filemanager-app (https://launchpad.net/~plasma-phone/+archive/ubuntu/ubuntu-apps)

bhush9, 2015-12-07:
Some animals in farm...

[image]

bhush9, 2016-01-05:
Alright, I got ubuntu-ui-toolkit built. making checklist of apps

- [x] animal-farm
- [x] calculator-app
- [x] sudoku-app
- [ ] samegame
- [ ] music app
- [ ] ubuntu-docviewer-app
- [x] ubuntu-terminal-app
- [ ] ubuntu-weather-app

more to follow, I will create new image with all working apps today.

star-buck, 2016-01-05:
@bhush9:
Your list is missing the most important ones like filemanager and browser from the top post. So while testing, its good to see some of the other working again and shipped to compare with our plasma ones, please add filemanager and browser to the list.

bhush9, 2016-01-06:
amended list, sorry missed to add them to list.

bhush9, 2016-01-08:
I am marking this as Ready for QA. mostly all apps included in next image 17 other then webbrowser-app, webbrowser-app still needs lots of recompiling that doesn't work as expected and terminal-app (terminal-app installs one package that conflicts with filemanager-app, but given we already have konsole I did keep filemanager-app)

star-buck, 2016-01-08:
split list of apps into working, installed, but not yet working and not working.

bhush9, 2016-01-08:
hmm, right I missed music-app

https://quickgit.kde.org/?p=mobster.git&a=commitdiff&h=a1c5f8cbb0960246db72c453bd2a7fdc27993870&hp=f1318910436f9c2f489386cd6af2c06bf2081886

This is packages I installed.

bhush9, 2016-01-08:
>Where do you keep the package-build instructions for ubuntu-filemanager-packages to appear here: http://mobile.neon.pangea.pub/pool/main/u/ubuntu-filemanager-app/

Nowhere, this build recipes and packaging is picked directly from launchpad, last month I added this functionality in our CI, so it can directly fetch source code from launchpad/bzr and build on our CI. 

http://mobile.kci.pangea.pub:8080/view/launchpad/

currently following projects are getting built.

http://anonscm.debian.org/cgit/pkg-kde/ci-tooling.git/tree/data/mci.json#n81

> How about building the others apps the same way too, so we have those packages in our own repository should we need them for different release etc.?

Here ubuntu-ui-toolkit, unity-action-api are more important packages because they depend on qt5.4 ABI for vivid, and we need to rebuild them against our qt5.5. Other then we don't really need to rebuild QML packages such as animal-farm/ubuntu-weather-app. They can just be installed like normal ubuntu apps.. its just not so useful maintenance overhead, so we should let launchpad do their job.. ;-) and rebuild only if we need.

star-buck, 2016-01-08:
okay, thanks for explanation!

bhush9, 2016-01-12:
As talked with Sho_ on IRC icons not showing up on launcher is mostly problem on ubuntu-apps side, they have non-standard way of declaring icons in .desktop file. Also, whatever Icon weather/document viewer app defines is not in package.. which is also wrong

```
13:36 <bshah> Sho_: as developer/maintainer of app launcher(s) is weather-app@30.png valid name for Icon entry?
13:36 <mgraesslin> awesome! My ananlysis was correct then
13:37 <Sho_> bshah: Define 'Icon entry'?
13:37 <bshah> Icon entry in .desktop file
13:37 <Sho_> The .png seems wrong
13:38 <bshah> Sho_: https://paste.kde.org/p98fqyix4 <-
13:38 <Sho_> But basically: Depends on whatever the fd.o icon theme spec says for icon names
13:39 <Sho_> ' If the name is an absolute path, the given file will be used. If the name is not an absolute path, the algorithm described in the Icon Theme Specification will be used to locate the icon. '
13:39 <Sho_> So it's OK if you make it an absolute path
13:39 <bshah> but this doesn't look like absoulute path? and this icon is not shipped... ubuntu things are strange
13:41 <Sho_> ok so the icon theme spec technically puts no restrictions on the icon file names
13:41 <Sho_> however the lookup algorithm contains talk of thinks like iconname.extension and it defines a whitelist of allowed extensions
13:42 <Sho_> so having the extension in the looked up icon name is against the spirit of the spec
13:42 <Sho_> in fact the described algorithm will look for weather-app@30.png.png
13:42 <Sho_> and weather-app@30.png.svg and weather-app@30.png.xpm
13:43 <Sho_> bottom line: looks invalid to me
13:44 <bshah> okay
```

bhush9, 2016-01-12:
Next image will contain _music-app_ 

I will next look into building liboxideqt/webbrowser-app. About samegame, package in ppa seems to be outdated and wouldn't install.

star-buck, 2016-02-13:
new Xenial IMG does not ship any, so unmarking all

bhush9, 2016-02-15:
Two things:

- https://launchpad.net/~ubuntu-touch-coreapps-drivers doesn't publish for xenial (yet)
- Some stuffs for example webbrowser-app depends on ubuntu libhybris's compat layer, which makes it 1000% impossible to use.

star-buck, 2016-02-23:
tested animal-farm from xenial ppa (see above): installs, comes up, no sound.
Will test more now.

bhush9, 2016-07-26:
Recent image have stallboard, weather, calculator, filemanager and animal farm installed, but some of them crashes because of bug in upstream code (ubuntu-ui-toolkit) with Qt 5.7.