blue-systems/plamo-initial-dev_DEPRECATED #227: make static_ci in mci work with neon git

jriddell, 2015-12-14:
to keep debian folks happy static_ci should use neon git repos instead of debian pkg-kde repos

I changed mci.json statuc_ci "type": "neon" and set  ci-tooling/lib/projects.rb to use it.

But the build still seems to use anonscm.debian.org
looking at build logs at http://mobile.kci.pangea.pub:8080/job/vivid_unstable_qt_qtxmlpatterns_src/

what am I missing?

apachelogger, 2015-12-15:
>what am I missing?
a test for the code you added

apachelogger, 2015-12-15:
and a test showing your problem

Updating vivid_unstable_qt_qtxmlpatterns_src
<?xml version='1.0' encoding='UTF-8'?>
<project>
  <actions/>
  <description/>
  <keepDependencies>
    false
  </keepDependencies>
  <properties>
  </properties>
  <scm class='org.jenkinsci.plugins.multiplescms.MultiSCM' plugin='multiple-scms@0.4'>
    <scms>
      <hudson.plugins.git.GitSCM plugin='git@2.3.5'>
        <configVersion>
          2
        </configVersion>
        <userRemoteConfigs>
          <hudson.plugins.git.UserRemoteConfig>
            <url>
              git://git.edinburghlinux.co.uk/qt/qtxmlpatterns
            </url>
          </hudson.plugins.git.UserRemoteConfig>
        </userRemoteConfigs>
        <branches>
          <hudson.plugins.git.BranchSpec>
            <name>
              */kubuntu_vivid_mobile
            </name>
          </hudson.plugins.git.BranchSpec>
        </branches>
        <doGenerateSubmoduleConfigurations>
          false
        </doGenerateSubmoduleConfigurations>
        <submoduleCfg class='list'/>
        <extensions>
          <hudson.plugins.git.extensions.impl.RelativeTargetDirectory>
            <relativeTargetDir>
              packaging
            </relativeTargetDir>
          </hudson.plugins.git.extensions.impl.RelativeTargetDirectory>
        </extensions>
      </hudson.plugins.git.GitSCM>
    </scms>
  </scm>
  <assignedNode>
    master
  </assignedNode>
  <canRoam>
    false
  </canRoam>
  <disabled>
    false
  </disabled>
  <blockBuildWhenDownstreamBuilding>
    true
  </blockBuildWhenDownstreamBuilding>
  <blockBuildWhenUpstreamBuilding>
    true
  </blockBuildWhenUpstreamBuilding>
  <triggers/>
  <concurrentBuild>
    false
  </concurrentBuild>
  <builders>
    <hudson.tasks.Shell>
      <command>
         export DIST=vivid export TYPE=unstable if [ ! -d source ]; then mkdir
        source echo
        'http://http.debian.net/debian/pool/main/q/qtxmlpatterns-opensource-src/qtxmlpatterns-opensource-src_5.5.1.orig.tar.xz' > source/url fi ~/tooling/kci/contain.rb ~/ci-tooling/kci/orig_sourcer.rb 
      </command>
    </hudson.tasks.Shell>
  </builders>
  <publishers>
    <hudson.tasks.ArtifactArchiver>
      <artifacts>
        build/*
      </artifacts>
      <allowEmptyArchive>
        false
      </allowEmptyArchive>
      <onlyIfSuccessful>
        false
      </onlyIfSuccessful>
      <fingerprint>
        false
      </fingerprint>
      <defaultExcludes>
        true
      </defaultExcludes>
    </hudson.tasks.ArtifactArchiver>
    <hudson.tasks.BuildTrigger>
      <childProjects>
        vivid_unstable_qt_qtxmlpatterns_bin_amd64,vivid_unstable_qt_qtxmlpatterns_bin_armhf
      </childProjects>
      <threshold>
        <name>
          UNSTABLE
        </name>
        <ordinal>
          1
        </ordinal>
        <color>
          YELLOW
        </color>
        <completeBuild>
          true
        </completeBuild>
      </threshold>
    </hudson.tasks.BuildTrigger>
    <hudson.plugins.ircbot.IrcPublisher plugin='ircbot@2.26'>
      <targets class='empty-list'/>
      <strategy>
        ANY_FAILURE
      </strategy>
      <notifyOnBuildStart>
        false
      </notifyOnBuildStart>
      <notifySuspects>
        false
      </notifySuspects>
      <notifyCulprits>
        false
      </notifyCulprits>
      <notifyFixers>
        false
      </notifyFixers>
      <notifyUpstreamCommitters>
        false
      </notifyUpstreamCommitters>
      <buildToChatNotifier class='hudson.plugins.im.build_notify.SummaryOnlyBuildToChatNotifier' plugin='instant-messaging@1.35'/>
      <matrixMultiplier>
        ONLY_CONFIGURATIONS
      </matrixMultiplier>
      <channels/>
    </hudson.plugins.ircbot.IrcPublisher>
    <hudson.plugins.ws__cleanup.WsCleanup plugin='ws-cleanup@0.26'>
      <patterns>
        <hudson.plugins.ws__cleanup.Pattern>
          <pattern>
            build/*
          </pattern>
          <type>
            INCLUDE
          </type>
        </hudson.plugins.ws__cleanup.Pattern>
      </patterns>
      <deleteDirs>
        true
      </deleteDirs>
      <skipWhenFailed>
        false
      </skipWhenFailed>
      <cleanWhenSuccess>
        true
      </cleanWhenSuccess>
      <cleanWhenUnstable>
        true
      </cleanWhenUnstable>
      <cleanWhenFailure>
        true
      </cleanWhenFailure>
      <cleanWhenNotBuilt>
        true
      </cleanWhenNotBuilt>
      <cleanWhenAborted>
        true
      </cleanWhenAborted>
      <notFailBuild>
        false
      </notFailBuild>
      <cleanupMatrixParent>
        false
      </cleanupMatrixParent>
      <externalDelete/>
    </hudson.plugins.ws__cleanup.WsCleanup>
  </publishers>
  <buildWrappers>
    <hudson.plugins.timestamper.TimestamperBuildWrapper plugin='timestamper@1.6.2'/>
    <hudson.plugins.ansicolor.AnsiColorBuildWrapper plugin='ansicolor@0.4.1'>
      <colorMapName>
        xterm
      </colorMapName>
    </hudson.plugins.ansicolor.AnsiColorBuildWrapper>
  </buildWrappers>
</project>

bhush9, 2015-12-18:
Now builds seem to be fetching stuff from neon git..

http://mobile.kci.pangea.pub:8080/job/vivid_unstable_qt_qtx11extras_src/2/console

closed 2015-12-18 by bhush9

star-buck, 2015-12-18:
@bhushan:
Can we use

http://mobile.neon.pangea.pub:8080/job/vivid_unstable_qt_qtx11extras_src/2/console

as url in the future, as kci will be deprecated.