blue-systems/plamo-initial-dev_DEPRECATED #229: [usb]: can no longer connect via mtp

star-buck, 2015-12-16:
maybe mtp isnt started, but dolphin only shows this:
[image]
clicking a second time on the Nexus 5 entry:
[image]

bhush9, 2015-12-18:
mtp-server seems to be crashing on mobile

```
Program received signal SIGSEGV, Segmentation fault.
0xb6e377c4 in std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
(gdb) bt
#0  0xb6e377c4 in std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#1  0x00047dea in std::_Rb_tree_iterator<std::pair<std::tuple<std::string, std::string> const, std::function<void (core::dbus::types::Variant const&)> > > std::_Rb_tree<std::tuple<std::string, std::string>, std::pair<std::tuple<std::string, std::string> const, std::function<void (core::dbus::types::Variant const&)> >, std::_Select1st<std::pair<std::tuple<std::string, std::string> const, std::function<void (core::dbus::types::Variant const&)> > >, std::less<std::tuple<std::string, std::string> >, std::allocator<std::pair<std::tuple<std::string, std::string> const, std::function<void (core::dbus::types::Variant const&)> > > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<std::tuple<std::string, std::string> const&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<std::tuple<std::string, std::string> const, std::function<void (core::dbus::types::Variant const&)> > >, std::piecewise_construct_t const&, std::tuple<std::tuple<std::string, std::string> const&>&&, std::tuple<>&&) ()
#2  0x0004c2c2 in std::shared_ptr<core::dbus::Property<core::UnityGreeter::Properties::IsActive> > core::dbus::Object::get_property<core::UnityGreeter::Properties::IsActive>() ()
#3  0x0004c5e8 in MtpDaemon::MtpDaemon(int) ()
#4  0x00037ee6 in main ()
```

and then goes defunct

star-buck, 2015-12-18:
Is this still the mtp-server that vishesh got to work previously?

bhush9, 2015-12-18:
yes, it is same mtp-server, forked on kde git.

star-buck, 2015-12-18:
So is mtp-server failing related to UnityGreeter and can we fix it?

bhush9, 2015-12-18:
On Fri, Dec 18, 2015 at 11:17 AM, star-buck <notifications@github.com>
wrote:

> So is mtp-server failing related to UnityGreeter and can we fix it?

Yes, I will look into fixing this. though since our mtp-server doesn't look
for UnityGreeter it sounds packaging problem or something.

-- 
Bhushan Shah

http://bhush9.github.io
IRC Nick : bshah on Freenode

bhush9, 2015-12-18:
http://commits.kde.org/mtp-server/c5e60c21d155b4960801bd7595a045103ae62892

There fixed, will be available in IMG14 scheduled today evening.

star-buck, 2015-12-18:
works again

closed 2015-12-18 by star-buck