blue-systems/plamo-initial-dev_DEPRECATED #232: [peruse]: doesnt start

star-buck, 2015-12-18:


star-buck, 2015-12-18:
konsole throws error message: 
Mobilemain.qml:97 Type Bookshelf unavailable

leinir, 2015-12-18:
There should be more errors than that, unless Bookshelf isn't packaged (which would be strange, as it's just a qml file installed alongside MobileMain.qml). I'm using multirom, so i can't check this error right now, but i shall test it when Bhushan pushes the next release. If you can get me the full output, though, i should be able to diagnose it more easily :)

bhush9, 2015-12-18:
I've really messed up multirom setups by enabling lockscreen by default.. I will be able to redo next image tomorrow morning. :-(

star-buck, 2015-12-18:
@bhush9 : if certain parts of the prototype influence specifically multirom, dont care for them. We all have a Nexus 5 fully to be used with PM only if necessary at this point. 

@leinir :
PM img not working on multirom is NOT an excuse to not test latest img.

bhush9, 2015-12-18:
@star-buck Yes, but well.. I personally use multirom based setup because it is more convenient to have two different images installed, one stable and one devel-proposed for testing purpose. Also lockscreen starting without setting password is bug that we will have to fix on lockscreen stack.

star-buck, 2015-12-18:
@bhush9 : yes, but we dont have to do that now. 1234 for non multirom is fine. also "stable" rom is as good or bad as non-stable at this point, unless devel-proposed is getting so much worse with each progression that its almost unusable... again, not testing devel-proposed because of multirom is NO excuse with everyone within blue systems having a nexus 5 exactly for testing devel-proposed.

star-buck, 2015-12-18:
@leinir : please test peruse on img14 yourself and see whats wrong.

leinir, 2015-12-19:
Right, turns out it was caused by a change in the mobile components. Fixed in 877024ee2335c5828bbf83b5a2224f0a1384368f, should be sorted next time the peruse package is updated - @bhush9 over to you :)

@star-buck Minor comment on the error output, the relevant log entry was the last error, which said that Categoryshelf.qml:33 Cannot assign to non-existent property "flickable" (just to make things easier another time an error like that happens, though hopefully the adjustment to my workflow to include updating my local mobile components more often should avoid this particular issue happening)

bhush9, 2015-12-19:
packaged. will be in next image..

star-buck, 2015-12-19:
it seems to start after upgrading all packages.
So what file format does it pick up and where do I have to put the files for Peruse to find them?

star-buck, 2015-12-19:
opening new ticket

closed 2015-12-19 by star-buck