blue-systems/plamo-initial-dev_DEPRECATED #235: [peruse]: empty comic list

star-buck, 2015-12-19:
What file formats does it pick up and where does one have to put the files for Peruse to find them?

leinir, 2015-12-29:
First last: One has to put the comics into either Documents or Downloads (more can be added if we want, but those seemed the most sensible - perhaps Pictures as well?).

The searched formats are currently just cbz OR cbr OR cb7 OR cbt OR cba OR chm OR djvu OR epub OR pdf if baloo is enabled (that's the baloo search string), or pdf or cbz if not (discovered by filesystem search and mimetype check). This will be equalised, and extended, and is really just a more or less random selection of items for now.

star-buck, 2015-12-30:
i tried pdf and cbz in those folders, and nada. Can you make a photo from another device to show how this looks on a Nexus 5?

leinir, 2015-12-31:
i certainly can - however, it turns out i had not, in fact, scheduled the build i thought i had (and then got away from it, hunting a crash). The build has now completed and i've just checked that i can, in fact, read both cbz and pdf comics on the device - https://twitter.com/leinir/status/682491483144994816 :) (the listing does show some odd layout issues, i'll sort those out)

star-buck, 2015-12-31:
So updating "peruse" should now install a working version?

star-buck, 2015-12-31:
i just updated to 0.0+git20151231.1145+15.04-0 still doesnt show any comics.

leinir, 2016-01-05:
It is odd that it should fail to show comics, i most certainly have it working here... If you could try and upgrade to the version i've just pushed (0.0+git20160105.0848+15.04) and run it from konsole, and then tell me if it says it's using Baloo or not, as that'll help me hunt down the issue a bit.

leinir, 2016-01-05:
So, with a bit of help from Bhushan, we have discovered baloo is in something of an odd state on the phone by default. While this itself will need fixing, i also need to handle that situation and fall back accordingly. I'll get on that :)

leinir, 2016-01-05:
Oh, the odd state: Baloo has indexing is enabled, which causes peruse to attempt to use Baloo. Baloo reports that it can't open the database if you run "balooctl status" on the command line, though, and consequently the same happens for the api, which causes it to fail to return any entries to peruse, which then ends up with no comics to show.

tl;dr compliant version: i need to handle baloo failing and not just handle baloo being turned off :)

leinir, 2016-01-05:
Right! Bhushah tells me the version currently available (that is, 0.0+git20160105.1522+15.04-0) works - still works for me as well, and given he had the same issue you did, i expect it should also work for you now :)

star-buck, 2016-01-05:
finally confirmed working! :+1:

closed 2016-01-05 by star-buck

leinir, 2016-01-05:
Awesomesauce! :D Very glad to hear it :) Interestingly, of course, the bug we're run into here is somewhat unique to plasma phone (basically, baloo is turned on, but the database is broken from start, which can be fixed by running balooctl disable && balooctl enable), so we now have a comic viewer which is more fault tolerant than it would have otherwise been.. so... not bad, really ;)