blue-systems/plamo-initial-dev_DEPRECATED #237: [document viewer]: doesnt start

star-buck, 2015-12-19:
img14

bhush9, 2015-12-20:
Seems to be missing git add.. https://quickgit.kde.org/?p=okular.git&a=commit&h=2b4486a00de34cf0102997db8bb88462d4959de6 OkularDrawer.qml file is missing.

notmart, 2015-12-22:
eew, my bad then :/
will fix as soon as i get home next couple of days

notmart, 2015-12-23:
repackaging

bhush9, 2016-01-01:
Seems to work now.

star-buck, 2016-01-08:
neon17: still doesnt start

bhush9, 2016-01-08:
problem above was fixed, this is new problem that pop up :p

bhush9, 2016-01-08:
erm, which document viewer did you start? there are two now?

star-buck, 2016-01-08:
ours, not ubuntu

llelectronics, 2016-01-08:
Turns out folderlistmodel qml plugin is missing. Installing it and the okular document viewer works.

bhush9, 2016-01-08:
will fix packaging.

bhush9, 2016-01-09:
adjusted packaging in : https://github.com/plasma-phone-packaging/okular/commit/6a171e760667d22a854335a19c1e76e80e988adb will be fixed in next image.

star-buck, 2016-02-26:
fixed

closed 2016-02-26 by star-buck