blue-systems/plamo-initial-dev_DEPRECATED #239: [animal farm]: no sound

star-buck, 2016-01-08:
seems the animals make no sound.

bhush9, 2016-01-08:
umm, works for me?

star-buck, 2016-01-08:
odd... no sound here :musical_score:

aleixpol, 2016-01-08:
Sound doesn't work here either.

bhush9, 2016-01-08:
notmart | bshah: bah, finally managed to reproduce the black battery thing on the aptop, hopefully i know what is it     
     bshah | getting it reproduced on laptop means almost fixed                                                             
  notmart_ | in that magic of going around looking for parents i only consider qquickitems, but i'm doing it from a qobject 
  notmart_ | so yeah, it loses a piece

bhush9, 2016-01-08:
err ignore

bhush9, 2016-01-08:
its really odd, it works for me.. but maybe I did upgrade and installed packages afterwards.. will reflash and check whats wrong.

bhush9, 2016-01-09:
hmm, missing dep : gstreamer1.0-plugins-ugly

install it and animals will make noise..

star-buck, 2016-01-09:
good to know.
will mark it solved and close after shipping in next img also works.

bhush9, 2016-01-12:
img18

closed 2016-01-12 by star-buck