blue-systems/plamo-initial-dev_DEPRECATED #242: [wifi]: after first successful connect, doesnt connect after reboot (status: "waiting for authorization forever")

star-buck, 2016-01-08:


star-buck, 2016-01-08:
i see the reason now:
popup opening in the background (behind dropped down toppanel) is asking for the password again.

mck182, 2016-01-08:
This is caused by kwallet timing out; bug introduced in kwallet commit 1d0bec904dd78ff21d1d68e9c3e363ad1a77f7ed.

I'm investigating.

mck182, 2016-01-08:
Fixed in kwallet; http://commits.kde.org/kwallet-framework/901c27ca7cb05cca1a960747b280d40cd7707158

bhush9, 2016-01-09:
Package available now : 5.18.0+git20160109.0454+15.04-0

star-buck, 2016-01-12:
it seems still not reconnecting without popup in img18...

star-buck, 2016-01-12:
okay, weird... it didnt for the first time when entering the password over the tiny field in the toppanel, but did so when entering the password a second time in the popup.

star-buck, 2016-02-26:
seems somewhat working, at least the popup comes up in background but wifi is established nevertheless.

closed 2016-02-26 by star-buck