blue-systems/plamo-initial-dev_DEPRECATED #243: libappstreamqt0 should be compiled against Qt5

aleixpol, 2016-01-08:
As of recently, Discover QApt backend depends on Appstream. The appstreamqt provided by Ubuntu Vivid depends on Qt 4 and Discover uses Qt5.

It would be really convenient if we could get the Qt5 version as otherwise we can't link against it.

To that end, we'd need a newer version, since the available one is 0.7.4 which is too old for us.

bhush9, 2016-01-13:
appstreamqt itself is recompiled now against qt5.5 however due to discover->plasma-discover rename it won't build.

star-buck, 2016-01-15:
any idea to fix this?

bhush9, 2016-01-15:
I've recompiled appstreamqt against our qt5. However currently kubuntu packagers are working on discover -> plasma-discover so it will need little time before I can build discover against new appstreamqt.

bhush9, 2016-01-19:
Now plasma-discover starts.

bhush9, 2016-03-02:
I am closing this, since this was required for vivid and in xenial it starts fine.

closed 2016-03-02 by bhush9