blue-systems/plamo-initial-dev_DEPRECATED #245: [Settings]: Accounts: module kaccounts failed to load

star-buck, 2016-01-12:


bhush9, 2016-01-12:
As reported in #209 installing qml-module-ubuntu-onlineaccounts will fix that, I will adjust packaging for plasma-settings and mark this as Ready for QA.

bhush9, 2016-01-12:
fixed in : https://github.com/plasma-phone-packaging/plasma-settings-packaging/commit/cefa78511d6d9a448ad800fa14ba10fa263191f1 waiting for rebuild.

bhush9, 2016-01-12:
plasma-settings (0.0.1ubuntu5+git20160112.1306+15.04-0) vivid;

bhush9, 2016-07-27:
Accounts module now loads itself, but one can't add new accounts

bhush9, 2016-08-05:
```
tp-qt 0.9.6.1 WARN: GetAll(AccountManager) failed with "org.freedesktop.DBus.Error.ServiceUnknown" : "The name org.freedesktop.Telepathy.AccountManager was not provided by any .service files" 
"google"
Looking for plugin ""
Starting auth session with "oauth2"
Info:
        Id: 3
        caption: "google"
        owner: ""
        userName: ""
"ktp-generic"
Looking for plugin ""
Starting auth session with "password"
Info:
        Id: 4
        caption: "ktp-generic"
        owner: ""
        userName: ""
qml: reacting to onCurrentModuleChanged 
Not loading plugin ... ""
```

some more debug info for record..

bhush9, 2016-08-05:
For start, I've removed all packages that I installed to get spacebar working.. To debug issue step by step, I am trying to get accounts import working first..

closed 2017-12-09 by star-buck