blue-systems/plamo-initial-dev_DEPRECATED #246: [wallpaper dialog]: pressing on empty area in homescreen brings up empty overlay

star-buck, 2016-01-12:
img18

notmart, 2016-01-12:
this seems related to another problem: that overlay is supposed to be there, with a toolbar at the bottom of the screen with "configure wallpaper" and "add widgets" buttons: however the panel geometries seems to be reported incorrectly and that toolbar ends up being hidden by the bottom panel, leaving that weird dark area

notmart, 2016-01-12:
latest plan is 9th february
https://wiki.qt.io/Qt-5.6-release

star-buck, 2016-01-15:
@notmart :
so qt5.6 will fix this? what part of qt is at fault here?

notmart, 2016-01-15:
ah, sorry. I had answered actually to the wrong issue, the answer was for the scrollwheel one.

the wrong struts was actually our problem and is fixed, updating the packages (or waiting next image) will. fix it

star-buck, 2016-02-13:
fixed with IMG20/xenial

closed 2016-02-13 by star-buck