blue-systems/plamo-initial-dev_DEPRECATED #248: [package request] Make build-dep for kaccounts* work

mck182, 2016-01-14:
sudo apt-get build-dep kaccounts-mobile would be very handy to get quickly all the stuff that is needed to work on that. Same for kaccounts-integration.

bhush9, 2016-01-15:
Hmm? whats preventing that? I don't understand.

mck182, 2016-01-15:
No clue.

$ sudo apt-get build-dep kaccounts-mobile

E: Unable to find a source package for kaccounts-mobile

(same with kaccounts-integration)

bhush9, 2016-01-15:
Ah, right, deb-src for mobile CI is not enabled in mobile IMG. I will make sure this gets corrected in next image.

bhush9, 2016-01-15:
Mean while please add,

```
deb-src http://mobile.kci.pangea.pub vivid main
```

to your sources.lst

bhush9, 2016-01-15:
fixed in : http://commits.kde.org/mobster/855e0713b5fc281cb4be4add86f1ff5b217e130a

mck182, 2016-01-15:
Awesome, thanks!

bhush9, 2016-01-15:
(also, I should replace mobile.kci with mobile.neon but that will require updating in all other places I believe)

mck182, 2016-01-15:
> Mean while please add,

It's there already, just commented out btw.

bhush9, 2016-01-20:
The way I added it didn't work out.. will need to find another way...

bhush9, 2016-07-27:
now that we are using live-build to create images, this is fixed, next week's image should have this fixed..

bhush9, 2016-08-02:
Fixed in image published today..

closed 2017-12-09 by star-buck