blue-systems/plamo-initial-dev_DEPRECATED #250: kaccounts + spacebar todo

bhush9, 2016-02-24:
Creating ticket for tracking

----------------------------------------------

Accounts
===========

- [ ] libaccounts-glib >= 1.21 (mostly master) https://gitlab.com/accounts-sso/libaccounts-glib/tags/VERSION_1.21
- [ ] kaccounts-integration from master
- [ ] kaccounts-providers from master
- [ ]  kaccounts-mobile from master

Spacebar
=============

- [ ] kpeople from master
- [ ] ktp-common-internals from mklapetek/SpaceBar
- [ ] spacebar from master
- [ ] plasma-phone-components from master

bhush9, 2016-02-25:
I've packed all things.. contact importing works but spacebar shows all 3 options about chat disabled.

mck182, 2016-02-25:
I forgot one package - telepathy-accounts-signon - https://gitlab.com/accounts-sso/telepathy-accounts-signon

bhush9, 2016-02-25:
telepathy-accounts-signon already packaged as mentioned in issue #247

bhush9, 2016-02-25:
actually nvm.. build was added but didn't build :P

bhush9, 2016-02-25:
Erm... no.. package is there.. false alarm

mck182, 2016-02-25:
So more packages need to be installed, namely these:
kde-config-telepathy-accounts
kde-telepathy-auth-handler
kde-telepathy-kaccounts
kde-telepathy-kpeople
libktpcommoninternals9
libktplogger9
libktpmodels9
qml-module-org-kde-telepathy

There are however some serious issues on the phone wrt Telepathy, preventing SpaceBar to work properly. Most importantly, telepathy-ofono breaks the presence controls, making it unable to connect at all (in short: ofono does not understand presence "available", so setting all accounts to available fails on ofono and so all accounts are set as offline).

Do we ship any patches in our mission-control package? There should be one regarding ofono. If we indeed ship that one, I'll have to figure out a way to make the ofono account ignored by presence controls.

For the other issues I've created new column in phrabrikator to track all mobile issues.

bhush9, 2016-02-26:
> So more packages need to be installed, namely these:
> kde-config-telepathy-accounts
> kde-telepathy-auth-handler
> kde-telepathy-kaccounts
> kde-telepathy-kpeople
> libktpcommoninternals9
> libktplogger9
> libktpmodels9
> qml-module-org-kde-telepathy
> 

Last night I already added `qml-module-org-kde-telepathy` to the deplist of spacebar. and I believe others are being pulled by them, still will verify.

> Do we ship any patches in our mission-control package? There should be one regarding ofono. If we indeed ship that one, I'll have to figure out a way to make the ofono account ignored by presence controls.

I think in xenial mobile IMG we don't ship any patches.. I believe they were required in vivid because they were already up-streamed but ubuntu packages were very old?

mck182, 2016-02-26:
> I think in xenial mobile IMG we don't ship any patches.. I believe they were required in vivid because they were already up-streamed but ubuntu packages were very old?

They were certainly not applied upstream yet [1], there seems to be nobody active these days. As for xenial package, I have no clue.

[1] - https://bugs.freedesktop.org/show_bug.cgi?id=91272

bhush9, 2016-03-02:
> They were certainly not applied upstream yet [1], there seems to be nobody active these days. As for xenial package, I have no clue.

> [1] - https://bugs.freedesktop.org/show_bug.cgi?id=91272

https://bugs.launchpad.net/ubuntu/+source/telepathy-mission-control-5/+bug/1473011

says it is in for xenial.

mck182, 2016-03-02:
> says it is in for xenial.

All I see is "telepathy-mission-control-5 (1:5.16.3-1ubuntu4) wily;" in that link? So I guess the xenial package is built from the wily one?

bhush9, 2016-03-03:
> So I guess the xenial package is built from the wily one?

yep, http://archive.ubuntu.com/ubuntu/pool/universe/t/telepathy-mission-control-5/telepathy-mission-control-5_5.16.3-1ubuntu6.debian.tar.xz from http://packages.ubuntu.com/xenial/telepathy-mission-control-5 contains that patch

bhush9, 2016-03-03:
I've added required apparmor permission stuff in telepathy-mission-control-5 package.

bhush9, 2016-03-03:
@mck182 it is shipped in IMG28 however still spacebar have all option disabled.. please check and let me know what changes are needed.

mck182, 2016-03-03:
If you're talking about the sidebar options, that's fine, they are not meant to be enabled yet.

bhush9, 2016-03-03:
Ah..

bhush9, 2016-03-03:
So can I assume spacebar is functional?

mck182, 2016-03-03:
I guess. The question is if Telepathy is functional. I'll try flashing right now.

mck182, 2016-03-03:
Ok, one thing I forgot to mention - the list of additional ktp packages _must_ not contain these:
kde-telepathy-text-ui
kde-telepathy-contact-list
kde-telepathy-approver

That said, SpaceBar still has troubles, but getting there.

closed 2017-12-09 by star-buck