blue-systems/plamo-initial-dev_DEPRECATED #252: [gnome-chess]: fails to start with error "the name a11y.Bus was not provided by any .service files"

star-buck, 2016-02-26:


bhush9, 2016-02-26:
Quick google search leads to some crazy apparmor thingie.

star-buck, 2016-02-26:
try if you can fix it next week

star-buck, 2016-07-26:
@bhush9 : status?

bhush9, 2016-07-26:
This seems to be upstream issue, that gnome-chess can't be run with wayland.. now it throws another error

(gnome-chess:3800): Gdk-CRITICAL **: /build/gtk+3.0-0Gk42k/gtk+3.0-3.18.9/./gdk/wayland/gdkdisplay-wayland.c:993: creating shared memory file failed: Operation not permitted

Which might be related with xdg-shell related changes in kwin..

star-buck, 2016-07-26:
it worked fine under (x)wayland on pm before?

bhush9, 2016-07-26:
At some point yes.. Issue mentioned in ticket title seems to be some other unrelated issue, googling provides results of 2012... and as it says, its just warning.

bhush9, 2016-07-27:
I just realized that doing export GDK_BACKEND=x11 before starting gnome-chess makes it start.. I've two options,

a) Add that to /etc/environment
b) Maybe @mgraesslin have any idea about shared memory file error and might be possible bug in kwin's xdg_shell implementation.

bhush9, 2016-07-27:
Personally, I'd advertise against option a) because it will prevent every GTK application to start in wayland mode.

mgraesslin, 2016-08-01:
> have any idea about shared memory file error and might be possible bug in kwin's xdg_shell implementation.

No, that has nothing to do with xdg_shell. Shared memory is the wl_shm interface and that works in general. Given the warning about the Operation not permitted I think the problem is with apparmor. GTK is not allowed to create the shared-memory-file. In opposite Qt seems to be able to create such files as KWrite works on the phone.

bhush9, 2016-08-01:
I would be surprised if it is apparmor issue, given in our kernel apparmor patches are not backported at all..

bhush9, 2016-08-01:
So... I guess I should export GDK_BACKEND=x11 on phone then...

mgraesslin, 2016-08-01:
> So... I guess I should export GDK_BACKEND=x11 on phone then...

No, we should try to figure out where GTK tries to create the shm file and allow it in apparmor (or bug Ubuntu for having an apparmor rule breaking GTK/Wayland).

bhush9, 2016-08-01:
Doing strace shows following,

```
memfd_create("gdk-wayland", MFD_CLOEXEC) = -1 ENOSYS (Function not implemented)
brk(0x73d000)                           = 0x73d000
write(2, "\n(gnome-chess:3898): Gdk-CRITICA"..., 175) = 175
mmap2(NULL, 1294336, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb1d76000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0} ---
+++ killed by SIGSEGV +++
```

memfd_create syscall was introduced in the kernel 3.18.. while on phone we have 3.4 kernel.. so it sounds like there is no possibility of having wayland with gtk on phone..

bhush9, 2016-08-01:
Enabled in https://github.com/plasma-phone-packaging/live-config/commit/c0ad1cb8a8cc3c5fa1c369eb525661b47146f42b : will be included in next image..

bhush9, 2016-08-02:
Fixed in the last image..

closed 2017-12-09 by star-buck