blue-systems/plamo-initial-dev_DEPRECATED #254: [animal farm]: exist whenever an animal is pressed and supposed to issue a sound

star-buck, 2016-07-26:


bhush9, 2016-07-26:
As mentioned in other ticket this issue is ubuntu-ui-toolkit being incompatible with Qt5.7 or having bug. I don't really have option but wait till ubuntu-ui-toolkit fixes this.. It seems to use private property of Qt5.6 which is not available in Qt5.7 from error it looks to me..

star-buck, 2016-07-26:
any other idea? it seems sound related, also stallboard laods and runs fine until pressing a tile...

bhush9, 2016-07-26:
No, it is not related to sound, just incompatibility with Qt5.7. This broke when we upgraded to qt5.7

star-buck, 2016-07-26:
but then it still works fine _until_ some sound should come up, so its not completely broken

bhush9, 2016-07-26:
Yes, in fact I believe it is related to animation it plays while touching animal

bhush9, 2016-07-28:
Warnings in console,

```
file:///usr/lib/arm-linux-gnueabihf/qt5/qml/Ubuntu/Components/1.2/MainView.qml:188: TypeError: Cannot read property '__tabBar' of null
file:///usr/lib/arm-linux-gnueabihf/qt5/qml/Ubuntu/Components/1.2/MainView.qml:188: TypeError: Cannot read property '__tabBar' of null
```

And backtrace of crash,

```
0xb62464b8 in QMetaObject::cast(QObject const*) const () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
(gdb) bt
#0  0xb62464b8 in QMetaObject::cast(QObject const*) const () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#1  0xaffd47e0 in qobject_cast<UCStyledItemBase*> (object=0xbf8d50) at /usr/include/arm-linux-gnueabihf/qt5/QtCore/qobject.h:518
#2  UCStyledItemBasePrivate::isParentFocusable (this=<optimized out>) at plugin/ucstyleditembase.cpp:118
#3  0x00cdbb30 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

bhush9, 2016-07-28:
I am going to try rebuild ubuntu-ui-toolkit and see if this gets fixed.. lets see

bhush9, 2016-07-28:
Sadly, build is still broken for armhf : http://mobile.neon.pangea.pub:8080/job/xenial_unstable_launchpad_ubuntu-ui-toolkit_bin_armhf/5/console

So I ca't upgrade ubuntu-ui-toolkit.. will think of another solutions if possible

aleixpol, 2016-07-28:
Maybe it would be possible to generate full Snap packages out of the click ones, containing the dependencies they might have? This way it would be possible to have Qt 5.4 (?) for the application while having a different Qt for the whole system.

bhush9, 2016-07-28:
Not sure how big snap package would be compared to current deb package with few KBs.. overall I would avoid that due to space constraints..

star-buck, 2016-07-28:
Still could be interesting to test.

Snaps will likely play a part in canonicals phone plans anyway, also
current similar apks are also working, so i dont see constraint space so
much of a limitation if things are not getting totally out of proportion.

bhush9, 2016-07-28:
We can test it for sure, but I wanted to avoid including it in default image.. mainly because current image itself is 400+ MB.

aleixpol, 2016-07-28:
> Not sure how big snap package would be compared to current deb package with few KBs.. overall I would avoid that due to space constraints..

Admittedly, it's a long shot. Ubuntu SDK and Desktop teams haven't adopted Snappy yet. OTOH, it's something that might be useful some day. Figuring out how to compile ubuntu-ui-toolkit can easily take a week to then realize it's not possible or having it break down the line again.

star-buck, 2018-02-08:
@bhush9 : since we use this nice little videoplayer from leszek on Plamo (which seems gotten an update):
https://openrepos.net/content/llelectronics/lls-video-player

please see if you can package this version of animal farm from here and try if it runs: https://openrepos.net/content/kuba77/sail-animal-farm

aleixpol, 2018-02-09:
I doubt it's doable, many sailfish API's are closed-source.

llelectronics, 2018-02-09:
Needs a simple port. My video player is also a port where I just use some underlying logic from the Sailfish version but rewrote the whole UI + someone ported that UI to Kirigami even and I gladly accepted this code. As the Sailfish UI is so unique it also would not make much sense to have it directly running on Plasma Mobile as it would feel totally out of place.