blue-systems/plamo-initial-dev_DEPRECATED #256: [wifi]: says connected, but e.g. firefox or discover show no webcontent

star-buck, 2017-01-10:


bhush9, 2017-01-11:
This seems to be due to SSL errors, installing ntp package fixes it, I will install it on default image.

bhush9, 2017-01-13:
This is quite strange issue.. half of time it works and half time it doesn't (even with ntp installed and correct time set).. investigating.

bhush9, 2017-01-18:
This should be fixed for Nexus 5..

bhush9, 2017-12-10:
This is now fixed? (there is Discover crash right now soon after starting but that's different issue)

star-buck, 2017-12-10:
Well, firefox crashing really prevents from knowing its fixed, but i assume
so.

Other than that, can you (or anyone else) confirm networkmanager doesnt
accept wifi password upon first entering it in the applet, even if
correctly and pops up to ask a second time?
Seems like this for latest mobile as well as on desktop, e.g. pinebook.

bhush9, 2017-12-10:
> Other than that, can you (or anyone else) confirm networkmanager doesnt accept wifi password upon first entering it in the applet, even if correctly and pops up to ask a second time?

I don't see this on current mobile image.. that said there was indeed a bug in plasma-nm which got fixed very recently... it might be that that bug fix is not released with latest plasma-nm version on pinebook.. but for mobile it works.

closed 2017-12-10 by star-buck