blue-systems/plamo-initial-dev_DEPRECATED #258: flashing nexus5x no longer finishes

star-buck, 2017-02-18:


star-buck, 2017-02-18:
its showing "waiting for device" forever

aleixpol, 2017-02-19:
I've had that before (with the nexus5), flashing with sudo helped, although it basically means that your system didn't recognize the nexus5x as a device you can temper with.

bhush9, 2017-02-20:
is it at fastboot stage? (i.e very start of script?) or later when there are calls to adb wait-for-device?

star-buck, 2017-02-20:
Later, device doesnt show up...

bhush9, 2017-02-20:
Okay, so its case of adb wait-for-device thingie. Someone on telegram also had same issue, and he earlier provided me patch that worked for him.. I don't know why this never fails for me though :man_shrugging: 

https://github.com/plasma-phone-packaging/pm-flashtool/commit/e2ebd2aa3020440088e374a12d4fb5d4e9ce7f59

Can you try this?

star-buck, 2017-02-20:
Please fix it so it works for all like it did before here.

bhush9, 2017-02-20:
Yes, this commit above is supposed to fix it..

notmart, 2017-02-20:
i just flashed the 5x with master from pm-flashtool, worked out of the box

bhush9, 2017-02-20:
Yeah, device with plasma mobile on it already have no problem.. its just devices with android as @star-buck's new phone is..

bhush9, 2017-02-22:
We can close this ticket I assume?

closed 2017-12-09 by star-buck