blue-systems/plamo-initial-dev_DEPRECATED #259: datetime: reads 1970 again

star-buck, 2017-02-22:


bhush9, 2017-02-22:
Are you connected to network? If so it would attempt to sync time from the ntp server..

star-buck, 2017-02-22:
we had this working with a recent date before without ntp, so its a regression.

bhush9, 2017-12-14:
This issue is fixed locally, marking it as needs shipping as image is left to publish.

bhush9, 2017-12-14:
Newer image `20171214-165756` should fix at least 1970. However solution in this image is missing one last bit which makes it show exact correct time and date.. I will include it in next image.

bhush9, 2017-12-19:
This is fixed properly in latest image. please test.

bhush9, 2017-12-20:
https://github.com/plasma-phone-packaging/pm-flashtool/commit/7b40d3d6a0d9c3ddb88a811184f2f7ed5d6a634c you will also need this commit in flashtool... appearantly i forgot to push this fix earlier

star-buck, 2018-02-08:
now it reads February 15th 2018 on todays date ;)

closed 2018-02-08 by star-buck