blue-systems/plamo-initial-dev_DEPRECATED #262: List of installed apps (keep ticket open)

star-buck, 2017-12-09:
[ ] Phone
[ ] Camera
[ ] Chess
[ ] Chromium
[ ] Discover
[ ] Firefox
[ ] Kirigami Gallery
[ ] Koko
[ ] Konsole
[ ] Marble Maps
[ ] Peruse
[ ] Settings
[ ] SpaceBar
[ ] Take Screenshot
[ ] Touch Test
[ ] Touch Test (double?)
[ ] vPlayer

star-buck, 2017-12-09:
remove one of the double Touch Test entries?

bhush9, 2017-12-10:
There is just one Touch Test installed however it somehow shows up as double entry.. and I am not quite sure why.. that said Touch test application is now not required, it was added for debugging at Randa sprint, I will just remove it.

bhush9, 2017-12-12:
A small change in the app ist entries

I will temporary not include the discover for few upcoming images, as they are breaking the image build, as soon as discover packaging is fixed in the neon.. Creating seperate issue for that..

https://github.com/blue-systems/pm-neon/issues/269