blue-systems/plamo-initial-dev_DEPRECATED #263: KILL KWALLET!!!!

star-buck, 2017-12-09:
Its stupidlz annoying as hell, please make it never appear or remove it

bhush9, 2017-12-10:
It's packaged, now I am waiting for kio rebuild on neon CI to finish so I can build image (not related to the kwallet problem), will mark it as Ready for QA when done

star-buck, 2017-12-10:
Thanks a ton for getting this done!

I cant even imagine how to present PM with this thing coming up all the
time and preventing anything remotely useful like opening a website with
chromium etc. :/

star-buck, 2017-12-10:
I assume its solved via empty password?

bhush9, 2017-12-10:
Yes.. I'll include pre-configured kwallet so it won't ask for password, unfortunately I can't build image today, so tomorrow morning it will be

aleixpol, 2017-12-11:
Probably the right way to do it (and I guess how distros are configuring it nowadays) is by using it tied to PAM.

A quick search to show what I mean, unsure what's the right documentation ATM. https://www.dennogumi.org/2014/04/unlocking-kwallet-with-pam/

bhush9, 2017-12-11:
kwallet-pam won't work with phone, see https://github.com/blue-systems/pm-neon/issues/163#issuecomment-143150406

bhush9, 2017-12-11:
Fixed in `20171211-055004`

notmart, 2017-12-11:
> kwallet-pam won't work with phone, see #163 (comment)
the lock screen can't be used in any way to do that?

bhush9, 2017-12-11:
No, when user enters password in lockscreen, it's too late and first kwallet dialog is already in background.. there was patch my the comment I pointed above mentions exactly that issue

closed 2017-12-11 by star-buck