blue-systems/plamo-initial-dev_DEPRECATED #265: audio on nexus 5x (bullhead)

bhush9, 2017-12-10:
- [x] new system img with audio_policy.conf
- [ ] new rootfs with pulseaudio configuration

bhush9, 2017-12-10:
This has been done now. (rootfs version : `20171210-140255`)

star-buck, 2017-12-10:
I can reflash latest img to get sound now?

bhush9, 2017-12-10:
Yes, you can..  at least oh it's gorgeous works now! ;-)

bhush9, 2017-12-10:
I'll experiment on currently available music players once issue #264 is resolved, there are some stability issues I want to focus on first.

bhush9, 2017-12-11:
just as FYI, there was bug in flash script, which made it re-use the old system.img if present, it is fixed in latest git master : https://github.com/plasma-phone-packaging/pm-flashtool/commit/75e195b27da3cad8bb0e4f9ea4a545dfb1094d0e

closed 2017-12-12 by star-buck