blue-systems/plamo-initial-dev_DEPRECATED #266: firefox: crashing upon startup

star-buck, 2017-12-10:


star-buck, 2017-12-10:
Btw., do we already package firefox quantum?

bhush9, 2017-12-11:
I'll need some help on this one, I believe since introduction of xdg shell in kwin_wayland firefox seems to crash on mobile, even if it is using Xwayland.. I don't know how to collect debug information so can't do much here.

Do @llelectronics or @shadeslayer know how to get debug log for firefox as they have dealt with firefox earlier.

bhush9, 2017-12-12:
Forgot to mention, but:

> Btw., do we already package firefox quantum?
Yes, the firefox in image seems to be firefox quantum already (based on icon)

star-buck, 2017-12-13:
would be pretty nice to have ff not crashing

bhush9, 2017-12-19:
downgrading firefox "fixes" it, but would be nice to debug it first

```
sudo apt install firefox=45.0.2+build1-0ubuntu1
```

bhush9, 2017-12-19:
So this seems to be "regression" from firefox side, yet to find out what is bug but I also tried firefox-esr from ppa and it seems to work, I think till bug is fixed we can ship firefox-esr?

bhush9, 2017-12-19:
To get it working locally

```
sudo apt install software-properties-common
sudo add-apt-repository ppa:mozillateam/ppa
sudo apt-get update
sudo apt install firefox-esr
```

star-buck, 2017-12-19:
We want ff quantum for several reasons, primary is because startup is much
faster compared to old versions.

The question is why doesnt it start and if related to wayland or anything
only happening on plasma side, we need to fix it for wayland desktop plasma
anyway.

bhush9, 2017-12-19:
That is regression from firefox side, I am talking with people in firefox IRC network and trying to pin-point it. It was false to assume that problem was on our side.

bhush9, 2017-12-19:
(To be more specific there is regression which affects on ARM platform)

star-buck, 2017-12-19:
Armhf? Because it works on the pinebook....

bhush9, 2017-12-19:
We've identified possibly related bug report to be : https://bugzilla.mozilla.org/show_bug.cgi?id=1306746 which is mostly reported on armhf / android devices, though we are not sure as I can't get useful backtraces yet.. currently information crash itself is really vague but at least surely is not issue on side of Plasma.

star-buck, 2018-02-08:
@bhush9 : please look into this with priority to get fixed with FF Quantum.

bhush9, 2018-02-08:
Copy pasting from IRC: 

I am afraid I can't do much, this is regression from upstream and I've reported them the crasher with the backtrace but otherwise I haven't heard back from them