blue-systems/plamo-initial-dev_DEPRECATED #267: Flashing latest img never comes up, stuck at google startscreen :/

star-buck, 2017-12-11:
log:
`+ CACHEDIR=cache
+ echo 'Waiting for device to be in the fastboot mode'
Waiting for device to be in the fastboot mode
+ fastboot getvar product
< waiting for any device >
product: bullhead
finished. total time: 0.080s
++ fastboot getvar product
++ head -1
++ awk '-F: ' '{print $2}'
+ DEVICE_NAME=bullhead
+ confirm 'Connected device is bullhead, is that correct? [y/N]'
+ read -r -p 'Connected device is bullhead, is that correct? [y/N] ' response
Connected device is bullhead, is that correct? [y/N] y
+ case $response in
+ true
+ PLATFORM=neon
+ echo 'Continuing ...'
Continuing ...
+ getopts cp: opt
+ case $opt in
+ echo 'Option provided to use cache, not downloading files again'
Option provided to use cache, not downloading files again
+ NOCACHE=1
+ getopts cp: opt
+ '[' -z 1 ']'
+ flash-phone bullhead
+ '[' 1 -eq 0 ']'
+ echo -n 'Flashing recovery image ... '
Flashing recovery image ... + fastboot flash recovery cache/bullhead/recovery.img
+ echo -n '[done]'
[done]+ fastboot boot cache/bullhead/recovery.img
downloading 'boot.img'...
OKAY [  0.603s]
booting...
OKAY [  0.547s]
finished. total time: 1.150s
+ wait_for_device
++ adb shell echo 1
+ test -z ''
+ echo -n .
.+ sleep 3
++ adb shell echo 1
+ test -z ''
+ echo -n .
.+ sleep 3
++ adb shell echo 1
+ test -z $'1\r'
+ echo

++ readlink -f cache/pm-rootfs-latest.tar.gz
+ ./rootstock-touch-install /home/pine64/pm-flashtool/cache/rootfs/pm-rootfs-20171211-112034.tar.gz cache/bullhead/latest/system.img cache/bullhead/latest/boot.img
mount: mounting /dev/block/mmcblk0p40 on /cache failed: Device or resource busy
mount: mounting /dev/block/mmcblk0p45 on /data failed: Device or resource busy
transfering rootfs tarball ... [done]
preparing system-image on device ... [done]
unpacking rootfs tarball to system-image ... [done]
adding android system image to installation ... __bionic_open_tzdata: couldn't find any tzdata when looking for GMT!
drwx------    2 root     root          4096 Aug  3  1970 adb
drwxrwxr-x    2 system   system        4096 Feb 11  2016 anr
drwxrwx--x    2 system   system        4096 Aug  3  1970 app
drwx------    2 root     root          4096 Aug  3  1970 app-asec
drwxrwx--x    2 system   system        4096 Feb 11  2016 app-ephemeral
drwxrwx--x    2 system   system        4096 Aug  3  1970 app-lib
drwxrwx--x    2 system   system        4096 Aug  3  1970 app-private
drwx------    2 system   system        4096 Aug  3  1970 backup
drwxr-xr-x    2 shell    shell         4096 Aug  3  1970 bootchart
lrwxrwxrwx    1 root     root            45 Aug  3  1970 bugreports -> /data/data/com.android.shell/files/bugreports
drwxrwx---    5 system   cache         4096 Feb 11  2016 cache
drwxrwx--x    2 system   system        4096 Aug  3  1970 connectivity
drwxrwx--x    5 root     root          4096 Aug  3  1970 dalvik-cache
drwxrwx--x    2 system   system        4096 Aug  3  1970 data
drwxrwxrwx    2 system   system        4096 Aug  3  1970 diag_logs
drwxrwx---    3 drm      drm           4096 Sep 24  1970 drm
drwxrwx---    2 system   system        4096 Aug  3  1970 fpc
drwxrwx---    2 system   system        4096 Aug  3  1970 fpc_tpl
drwxr-x--x    3 root     root          4096 Aug  3  1970 local
drwxrwx---    2 root     root          4096 Jan  1 00:00 lost+found
drwxr-xr-x    4 root     root          4096 Aug  3  1970 lxc
drwxrwx---    4 media_rw media_rw      4096 Aug  3  1970 media
drwxrwx---    2 mediadrm mediadrm      4096 Aug  3  1970 mediadrm
drwxrwx--t   34 system   misc          4096 Feb 11  2016 misc
drwxrwx--t    2 system   misc          4096 Feb 11  2016 misc_ce
drwxrwx--t    2 system   misc          4096 Feb 11  2016 misc_de
drwxrwx---    3 nfc      nfc           4096 Aug  3  1970 nfc
drwxrwx--x    2 root     root          4096 Feb 11  2016 ota
drwxrwx---    2 system   cache         4096 Feb 11  2016 ota_package
drwx------    2 root     root          4096 Feb 11  2016 property
drwx------    2 root     root          4096 Aug  3  1970 ramdump
drwxrwx--x    2 system   system        4096 Aug  3  1970 resource-cache
-rw-rw-rw-    1 root     root     2097152000 Jan 29 23:31 rootfs.img
drwx--x--x    2 system   system        4096 Aug  3  1970 security
drwx------    2 system   system        4096 Aug  3  1970 ss
drwxrwxr-x    4 system   system        4096 Feb 11  2016 system
-rw-rw-rw-    1 root     root     195399680 Dec 11  2017 system.img
drwxrwx---    2 system   system        4096 Feb 11  2016 system_ce
drwxrwx---    2 system   system        4096 Feb 11  2016 system_de
drwx------    2 system   system        4096 Aug  3  1970 time
drwxrwx--x    2 system   system        4096 Aug  3  1970 tombstones
drwx--x--x    2 system   system        4096 Aug  3  1970 user
drwx--x--x    2 system   system        4096 Feb 11  2016 user_de
[done]
cleaning up on device ... [done]
rebooting into bootloader mode and flashing boot.img ... [done]`

bhush9, 2017-12-11:
It seemed to work fine for me with multiple reflash whole day, bbut nevertheless I'll try to investigate tomorrow

star-buck, 2017-12-11:
Can you CONFIRM its not working on nexus5x??

bhush9, 2017-12-11:
A different issue I have seen is there is a slight delay of approximate ~30 second before UI starts.. maybe it's that?

star-buck, 2017-12-11:
CAN YOU TEST IT instead of asking questions???

bhush9, 2017-12-11:
I've tested it, that's what I mentioned in first comment.. and last notmart also flashed just fine.. at moment I've already gone off to bed so I am afraid I can't test it.. (11:50 pm, will test and report back tomorrow morning)

star-buck, 2017-12-11:
It is NOT working, 100% confirmed, also not after waiting 30 seconds or 3
minutes...
I dont care if you tested flashing any PREVIOUS imgs, as those were working
fine for me too.
I only care you tested the LAST img after you changed anything COMPLETE
redownloading and flashing from your own script, and that is broken, so
nothing else counts.

bhush9, 2017-12-11:
Okay, I just confirmed what is issue and have uploaded a fixed image (basically file corruption on server, sorry for that)

And FWIW I've flashed same image earlier but then had to check file on image server and in process file got corrupted

Issue should be resolved now..

closed 2017-12-11 by star-buck